atsamd-rs/atsamd #295: Wio Terminal UART does not work with external clock mode

tomoyuki-nakabayashi, 2020-10-28:
I'm attempting to use Wio Terminal BSP crate, things are looking good!
I notice that UART function does not work current master 1dd79c0.

When I uncomment the below line, i.e., use internal clock, it works well.

https://github.com/atsamd-rs/atsamd/blob/1dd79c0a0b86067d57b9a3cab25e890a3ce947f7/hal/src/common/thumbv7em/sercom/uart.rs#L159

From datasheet, the default value is external clock (USART) mode.
Ideally, it would be better to select UART / USART mode from application.
But for now, I'm not sure which mode is preferable for atsamd-hal implementation.

thanks in advance,

Here, my uart application code as a reference.

```rust
#![no_std]
#![no_main]

use panic_halt as _;
use wio_terminal as wio;

use wio::hal::clock::GenericClockController;
use wio::hal::time::Hertz;
use wio::pac::Peripherals;
use wio::prelude::*;
use wio::{entry, Pins, Sets};
use core::fmt::Write;

#[entry]
fn main() -> ! {
    let mut peripherals = Peripherals::take().unwrap();

    let mut clocks = GenericClockController::with_external_32kosc(
        peripherals.GCLK,
        &mut peripherals.MCLK,
        &mut peripherals.OSC32KCTRL,
        &mut peripherals.OSCCTRL,
        &mut peripherals.NVMCTRL,
    );

    let mut sets: Sets = Pins::new(peripherals.PORT).split();
    let mut serial = sets.uart.init(
        &mut clocks,
        Hertz(115200u32),
        peripherals.SERCOM2,
        &mut peripherals.MCLK,
        &mut sets.port);

        write!(&mut serial, "hello {}", "world").unwrap();

    loop {}
}
```

TDHolmes, 2020-11-02:
@jessebraham any reason that line was commented out?

jessebraham, 2020-11-02:
It looks like it was commented out in the previous `same54` implementation, so I'm guessing it just got erroneously copied over and missed in review. I didn't write any of the `same54` stuff so I can't answer as to why that was commented out.

PR #301 (2020-11-03): UART uses internal clock mode
fix #295 

It might be erroneously commented out in #271.
I did some tests on the ATSAMD51P19A (Wio Terminal).
closed 2020-11-03 by sajattack