atsamd-rs/atsamd #312: Consider adding strongly typed errors (clippy result_unit_err)

VictorKoenders, 2020-11-11:
Clippy recently introduced a new warning: https://rust-lang.github.io/rust-clippy/master/index.html#result_unit_err
They recommend replacing all `Result<x, ()>` with a strongly typed struct.

One of the things I ran into was with `impl OutputPin`: https://github.com/atsamd-rs/atsamd/blob/master/hal/src/common/thumbv6m/gpio.rs#L364
It has a comment with `// TODO: switch to ! when it’s stable`, however it seems like there has not been much progress on stabilizing the never type: https://github.com/rust-lang/rust/issues/35121

I'd suggest making helper structs:

```rust
use core::fmt;

#[derive(Debug)]
// TODO: switch to ! when it’s stable
// tracking issue: https://github.com/rust-lang/rust/issues/35120
pub enum PinOutputError {}

impl fmt::Display for PinOutputError {
    fn fmt(&self, _f: &mut fmt::Formatter<'_>) -> fmt::Result {
        match self {
            _ => Ok(()),
        }
    }
}

#[derive(Debug)]
// TODO: switch to ! when it’s stable
// tracking issue: https://github.com/rust-lang/rust/issues/35120
pub enum PinInputError {}

impl fmt::Display for PinInputError {
    fn fmt(&self, _f: &mut fmt::Formatter<'_>) -> fmt::Result {
        match self {
            _ => Ok(()),
        }
    }
}
```

and implementing these for `impl OutputPin` and `impl InputPin`

jessebraham, 2020-11-11:
There is work currently underway in #302 to entirely replace the GPIO module, just FYI.

bradleyharden, 2020-11-11:
I switched to `core::convert::Infallible` in the new GPIO version. Is there an advantage to the helper struct approach over that?

blm768, 2020-11-12:
> I switched to `core::convert::Infallible` in the new GPIO version. Is there an advantage to the helper struct approach over that?

Not that I'm aware. As I understand it, `core::convert::Infallible` is intended to eventually become a direct alias of the `!` type.

VictorKoenders, 2020-11-12:
`core::convert::Infallible` also works if the methods can never fail.

It would be nice if `embedded_hal` would support infallible input/output pins, but that's a completely different story

bradleyharden, 2020-11-20:
@VictorKoenders, now that #302 has been merged, has this been fixed for you?

VictorKoenders, 2020-11-20:
Yes I believe so. Thank you!

closed 2020-11-20 by VictorKoenders