atsamd-rs/atsamd #329: Refactor UART implementation

twitchyliquid64, 2020-11-23:
- [ ] Make UART generic over some trait that manages configuration of the baud rate (arithmetic vs fractional)
- [ ] Have methods to enable/disable interrupts & check intflag properly
- [ ] Handle errors properly by reading `intflag.ERR` + `status.*`, clearing them, and discarding bytes as appropriate
- [ ] Generated documentation needs work

niclashoyer, 2021-01-22:
Does this also include switching the number of data bits? I have a strange requirement where I need 9 data bits, no parity and one stop bit. While the SERCOM on some chips allow such a configuration, it is very uncommon and I feared that it will cause a lot of work down here.

TDHolmes, 2021-01-22:
That seems like a reasonable extension of "Make UART parameters configurable" to me

PR #443 (2021-07-31): UART v2
This PR aims to provide a new UART driver with an API similar to `sercom::v2::spi`.

In fact, most of the type-level code was shamelessly copied from @bradleyharden's SPI driver. Thanks for doing most of the legwork! 

# Supported
* Basic UART functionality including different frame configurations: parity, stop bits and character size are configurable.
* Fractional or Arithmetic baud mode
* Flow control (CTS/RTS pins). 
* Rx + Tx, Rx-only or Tx-only
* Splitting of Rx and Tx halves
* IrDA encoding

# Not supported
* LIN mode (thumbv7em devices)
* 32-bit extension mode (thumbv7em devices [... cut]
closed 2021-07-31 by bradleyharden