atsamd-rs/atsamd #333: Use embedded-time

jessebraham, 2020-11-24:
Remove [time.rs] in favour of [embedded-time]. This will require reworking any peripheral implementations which use frequency and/or time units.

Work on this is already underway in #308.

[time.rs]: https://github.com/atsamd-rs/atsamd/blob/master/hal/src/common/time.rs
[embedded-time]: https://github.com/FluenTech/embedded-time

twitchyliquid64, 2020-11-24:
Ive been working on this slowly with @jacobrosenthal in https://github.com/twitchyliquid64/atsamd/tree/EMBEDDED_TIEM.

One pain-point we are constantly running into the that embedded-time only converts `Hertz<u64>` into other types via the `From` trait, which is technically correct, but given we will never need a frequency > 4.1 Ghz this restriction just ends up being a pain.

I'm going to investigate ways to implement `From<Hertz<u32>>` so we can keep the existing behavior.

sajattack, 2020-11-24:
Ugh y they no TryFrom.

twitchyliquid64, 2020-11-24:
Theres `TryFrom`, but compared to just writing `400.Khz()` / `20.MHz()` etc everywhere, the `TryFrom` is unwieldy, not to mention we then need to import the `TryFrom` trait in each BSP crate.

sajattack, 2020-11-24:
Yeah it's not quite as clean but I think it makes sense to error if an invalid frequency is requested

tgross35, 2024-01-11:
I think this crate has switched to the better-supported [`fugit`](https://docs.rs/fugit/latest/fugit/), so can this issue be closed?

closed 2024-01-11 by jbeaurivage