atsamd-rs/atsamd #361: Adafruit metro m4 express airlift board support

robo-corg, 2020-12-19:
I got my hands on a [metro m4 express airlift](https://learn.adafruit.com/adafruit-metro-m4-express-airlift-wifi) with an embedded esp8266 that talks over sercom0 using wifi nina which there seems to be a crate for.

Would there be interest in a PR to add this board with the wifi pins defined? Besides the wifi co-processor it seems to be exactly the same as the metro_m4 already in this repo.

Given there appears to be a [wifi nina crate](https://crates.io/crates/wifi-nina) someone has already written, it would be cool to have an example using this or even a feature flag to directly integrate this into the board's crate.

sajattack, 2020-12-19:
I think it would be best implemented as an optional feature, similar to how the radios are implemented for feather m0.

bradleyharden, 2021-06-14:
@robo-corg, is that solution acceptable? Is there anything left to discuss? Or can we close this issue?

robo-corg, 2021-06-15:
Yup go for it. I can open a PR if and when I have the time (it has remapped
pins similar to other boards in this repo so it probably does warrant its
own board support).

closed 2021-06-15 by bradleyharden