atsamd-rs/atsamd #364: Aliasing GPIO names at the boar level still available for after v2 Refactor?

jaredwolff, 2020-12-28:
One thing I did notice during my migration here: #363  is whether or not now there's a way to alias the names of the pins for specific boards. Is that still possible or with the refactor that won't work anymore?

sajattack, 2020-12-29:
The define_pins macro is still available in v2 https://github.com/atsamd-rs/atsamd/blob/f36882efa0708126f59e995530c6420e43bb3da6/hal/src/common/gpio/v2/pin.rs#L1419

jaredwolff, 2020-12-29:
Thanks @sajattack 

Maybe it's my lack of understanding. Looks like that macro is being used here: https://github.com/atsamd-rs/atsamd/blob/f36882efa0708126f59e995530c6420e43bb3da6/hal/src/common/gpio/v2/pin.rs#L1447

Am I wrong?

I can access them this way. 

```rust
        let mut pins = Pins::new(peripherals.PORT);
        let led_pass = pins.pa23.into_push_pull_output();
```

sajattack, 2020-12-29:
I asked @bradleyharden about this and he said he hasn't implemented a solution for BSP aliases yet but he's going to work on it.

jaredwolff, 2020-12-29:
Cool. Thanks @sajattack. I'll close this out unless you guys want to keep this open for tracking purposes.

sajattack, 2020-12-29:
Might as well keep it open until it's implemented.

bradleyharden, 2021-01-02:
@jaredwolff, the `v2` `define_pins` macro is actually completely unrelated to the `v1` `define_pins` macro. That's just an unfortunate naming coincidence. The `v2` version is intended only for internal use.

I'm looking at the `v1` `define_pins` macro now, and I think it would make sense to define something equivalent for `v2`. I'll try to put something together that satisfies the same purpose and perhaps improves on it, as discussed in #363.

bradleyharden, 2021-01-23:
Does #371 satisfy this issue? If it is merged, should this issue be closed?

jaredwolff, 2021-01-23:
Correcto. We can close this out once that is merged.

PR #371 (2021-01-23): Fix gpio::v2 PORT access functions and add a bsp_pins macro
See discussion in #363.

@jaredwolff, could you give this a try and see what you like/don't like? I'd like to get some feedback on it before merging.

Edit: closes #364
closed 2021-01-23 by twitchyliquid64