atsamd-rs/atsamd #376: cargo-hf2: InvalidBinary

kdsch, 2021-01-09:
I am opening this issue at the request of @twitchyliquid64 (per our discussion [here](https://github.com/atsamd-rs/atsamd/issues/275#issuecomment-756950378)), but it pertains to an error encountered while running cargo-hf2, so let me know if this is not the right place to discuss it.

```
$ rustc --version
rustc 1.49.0 (e1884a8e3 2020-12-29)

$ cargo hf2 --version
cargo-hf2 0.3.1

$ uname -a
Linux XPS-13-9360 5.4.0-60-generic #67-Ubuntu SMP Tue Jan 5 18:31:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

$ cat .cargo/config.toml
[build]
target = "thumbv7em-none-eabihf"
```

I am new to Rust; I may have made a basic mistake. The project is to build Rust code that incorporates Faust (transpiled to Rust, incorporated via build.rs) to run on the Adafruit Grand Central M4. I'm using a bootloader. I have successfully flashed the blinky example for the board, and have used that project as a starting point of mine. The main difference is the directory structure (and anything I might have missed).

```
$ tree -L 2
.
├── build.rs
├── Cargo.lock
├── Cargo.toml
├── dev
├── memory.x
├── sp.dsp
├── src
│   └── main.rs
└── target
    ├── CACHEDIR.TAG
    ├── debug
    └── thumbv7em-none-eabihf

```

Here's the error.

```
cargo hf2 --target thumbv7em-none-eabihf
warning: function is never used: `faust`
  --> build.rs:13:4
   |
13 | fn faust(out_dir: &String) {
   |    ^^^^^
   |
   = note: `#[warn(dead_code)]` on by default

warning: 1 warning emitted

    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
    Searching for a connected device with known vid/pid pair.
    Trying  Ok(Some("Adafruit Industries")) Ok(Some("Grand Central M4 Express"))
    Flashing "/home/karl/rust/sp/target/thumbv7em-none-eabihf/debug/sp"
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: InvalidBinary', /home/karl/.cargo/registry/src/github.com-1ecc6299db9ec823/cargo-hf2-0.3.1/src/main.rs:118:47
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I inspected the cargo-hf2 source. Long story short, the ELF created by [invoking cargo](https://github.com/jacobrosenthal/hf2-rs/blob/349ddf6f598c06c61e9769f5a7fa2203acd042ae/cargo-hf2/src/main.rs#L59) is suspect.

This file at first seemed to be a valid ARM ELF

```
$ file target/thumbv7em-none-eabihf/debug/sp
target/thumbv7em-none-eabihf/debug/sp: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, with debug_info, not stripped
```
however upon closer inspection with Radare, has some unusual properties: no entrypoints, all symbols starting at zero and having zero length.

```
[0x00000000]> is
[Symbols]

nth paddr       vaddr      bind  type size lib name
―――――――――――――――――――――――――――――――――――――――――――――――――――
1    ---------- 0x00000000 LOCAL FILE 0        1incscjhn1ieeug0
2    ---------- 0x00000000 LOCAL FILE 0        1r860d6ez6ly5ofr
3    ---------- 0x00000000 LOCAL FILE 0        1u6we4a9zwsk1y7o
4    ---------- 0x00000000 LOCAL FILE 0        1x04659mx9gtddr
5    ---------- 0x00000000 LOCAL FILE 0        1y337l38dmn7rueh
6    ---------- 0x00000000 LOCAL FILE 0        1yuf4d0v7ayf0q1w
7    ---------- 0x00000000 LOCAL FILE 0        2fn1n747o8fcvq33
8    ---------- 0x00000000 LOCAL FILE 0        2qgxww5xms9ft1wu
9    ---------- 0x00000000 LOCAL FILE 0        2ynbogbx1os24l7g
10   ---------- 0x00000000 LOCAL FILE 0        30mexigz3wv5zs03
11   ---------- 0x00000000 LOCAL FILE 0        3i11jeqok4qeaqml
12   ---------- 0x00000000 LOCAL FILE 0        3t25h06mhn8fgv1v
13   ---------- 0x00000000 LOCAL FILE 0        4dm9qo27wl2giw0p
14   ---------- 0x00000000 LOCAL FILE 0        9v9p9jqlskaeulx
15   ---------- 0x00000000 LOCAL FILE 0        awmrnmghvn19q68
16   ---------- 0x00000000 LOCAL FILE 0        devzorvht4450g6
17   ---------- 0x00000000 LOCAL FILE 0        sm9zjk1ts9hq7y8
18   ---------- 0x00000000 LOCAL FILE 0        grand_central_m4.ay0vcyf4-cgu.0
19   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.0
20   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.10
21   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.11
22   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.13
23   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.14
24   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.12
25   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.15
26   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.4
27   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.7
28   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.8
29   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.5
30   ---------- 0x00000000 LOCAL FILE 0        atsamd_hal.9ugh0dcp-cgu.9
31   ---------- 0x00000000 LOCAL FILE 0        rand_core.bzf5ofjs-cgu.0
32   ---------- 0x00000000 LOCAL FILE 0        rand_core.bzf5ofjs-cgu.2
33   ---------- 0x00000000 LOCAL FILE 0        rand_core.bzf5ofjs-cgu.6
34   ---------- 0x00000000 LOCAL FILE 0        rand_core.bzf5ofjs-cgu.7
35   ---------- 0x00000000 LOCAL FILE 0        rand_core.bzf5ofjs-cgu.8
36   ---------- 0x00000000 LOCAL FILE 0        atsamd51p.bwgof76q-cgu.10
37   ---------- 0x00000000 LOCAL FILE 0        atsamd51p.bwgof76q-cgu.11
38   ---------- 0x00000000 LOCAL FILE 0        cortex_m_rt.7c5mh2pu-cgu.4
39   ---------- 0x00000000 LOCAL FILE 0        cortex_m_rt.7c5mh2pu-cgu.6
40   ---------- 0x00000000 LOCAL FILE 0        cortex_m_rt.7c5mh2pu-cgu.1
41   ---------- 0x00000000 LOCAL FILE 0        r0.87t0un7p-cgu.0
42   ---------- 0x00000000 LOCAL FILE 0        r0.87t0un7p-cgu.3
43   ---------- 0x00000000 LOCAL FILE 0        cortex_m.7bqin0p9-cgu.1
44   ---------- 0x00000000 LOCAL FILE 0        cortex_m.7bqin0p9-cgu.14
45   ---------- 0x00000000 LOCAL FILE 0        cortex_m.7bqin0p9-cgu.15
46   ---------- 0x00000000 LOCAL FILE 0        cortex_m.7bqin0p9-cgu.6
47   ---------- 0x00000000 LOCAL FILE 0        cortex_m.7bqin0p9-cgu.7
48   ---------- 0x00000000 LOCAL FILE 0        cortex_m.7bqin0p9-cgu.8
49   ---------- 0x00000000 LOCAL FILE 0        cortex_m.7bqin0p9-cgu.4
50   ---------- 0x00000000 LOCAL FILE 0        bare_metal.5bhsp06t-cgu.0
51   ---------- 0x00000000 LOCAL FILE 0        core.6jav0fw1-cgu.0
52   ---------- 0x00000000 LOCAL FILE 0        panic_halt.cs89rquf-cgu.0
53   ---------- 0x00000000 LOCAL FILE 0        panic_halt.cs89rquf-cgu.1
54   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.168
55   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.147
56   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.152
57   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.178
58   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.183
59   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.185
60   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.30
61   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.4
62   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.64
63   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.115
64   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.85
65   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.87
66   ---------- 0x00000000 LOCAL FILE 0        compiler_builtins.ap9uzz99-cgu.91
67   ---------- 0x00000000 LOCAL FILE 0        aeabi_div0.c

[0x00000000]> iS
[Sections]

nth paddr          size vaddr         vsize perm name
―――――――――――――――――――――――――――――――――――――――――――――――――――――
0   0x00000000      0x0 0x00000000      0x0 ---- 
1   0x00000094   0x3973 0x00000000   0x3973 ---- .debug_abbrev
2   0x00003a07  0x4f9a9 0x00000000  0x4f9a9 ---- .debug_info
3   0x000533b0   0x3d80 0x00000000   0x3d80 ---- .debug_aranges
4   0x00057130  0x18c50 0x00000000  0x18c50 ---- .debug_ranges
5   0x0006fd80  0x63cb7 0x00000000  0x63cb7 ---- .debug_str
6   0x000d3a37  0x1b387 0x00000000  0x1b387 ---- .debug_pubnames
7   0x000eedbe  0x24b31 0x00000000  0x24b31 ---- .debug_pubtypes
8   0x001138ef     0x3a 0x00000000     0x3a ---- .ARM.attributes
9   0x0011392c   0xb494 0x00000000   0xb494 ---- .debug_frame
10  0x0011edc0  0x328c8 0x00000000  0x328c8 ---- .debug_line
11  0x00151688     0xd7 0x00000000     0xd7 ---- .debug_loc
12  0x0015175f     0x93 0x00000000     0x93 ---- .comment
13  0x001517f4    0x440 0x00000000    0x440 ---- .symtab
14  0x00151c34     0xba 0x00000000     0xba ---- .shstrtab
15  0x00151cee    0x67a 0x00000000    0x67a ---- .strtab

[0x00000000]> ie
[Entrypoints]

0 entrypoints
```

jacobrosenthal, 2021-01-09:
Do you have your project on github? Are you using --nmagic flag?
https://github.com/atsamd-rs/atsamd/blob/master/boards/feather_m4/.cargo/config#L11

kdsch, 2021-01-09:
The code isn't on GitHub, but I can create a repo there if it would help.

Good suggestion. Looks like I _did_ miss the .cargo/config file. And that seems to have fixed the problem! Thank you.

closed 2021-01-09 by kdsch