atsamd-rs/atsamd #399: How to setup periodical timer interrupt on SeeedStudio Xiao(SAMD21)?

windoze, 2021-02-03:
I've tried following code but it doesn't seem to be working:
```
        let mut peripherals = Peripherals::take().unwrap();
        let mut clk_controller = GenericClockController::with_external_32kosc(
            peripherals.GCLK,
            &mut peripherals.PM,
            &mut peripherals.SYSCTRL,
            &mut peripherals.NVMCTRL,
        );
        let clock =clk_controller.gclk1();
        let tc2tc3clock = clk_controller.tcc2_tc3(&clock).unwrap();
        let mut t = TimerCounter::<TC3>::tc3_(&tc2tc3clock, peripherals.TC3, &mut peripherals.PM);
        t.enable_interrupt();
```
Am I doing anything wrong?

TDHolmes, 2021-02-03:
You are not starting the timer. Check out the `sleeping_timer` examples for some usages. 

You can also call `start` directly to kick off a count down timer, and then in the interrupt handler you can kick it back off by calling `start` again.

TDHolmes, 2021-04-10:
@windoze , any success?

bradleyharden, 2021-06-14:
@windoze, I'm going to close this. Please feel free to reopen if you have any more questions.

closed 2021-06-14 by bradleyharden

windoze, 2021-06-23:
Sorry for the late response.
I tried and it works, thanks!