atsamd-rs/atsamd #403: Gpio v2 ADC and USB support

jaredwolff, 2021-02-10:
I was having trouble using GPIO v2 with ADC and USB. Thanks to the note in my PR from @bradleyharden I was able to get USB working:

```
#[cfg(feature = "usb")]
pub fn usb_allocator(
    usb: pac::USB,
    clocks: &mut GenericClockController,
    pm: &mut pac::PM,
    dm: Pin<PA24, Input<Floating>>,
    dp: Pin<PA25, Input<Floating>>,
) -> UsbBusAllocator<UsbBus> {
    let gclk0 = clocks.gclk0();
    let usb_clock = &clocks.usb(&gclk0).unwrap();

    // Convert to v1
    let dm: Pa24<_> = dm.into_alternate().into();
    let dp: Pa25<_> = dp.into_alternate().into();

    // Then pop those into UsbBus
    UsbBusAllocator::new(UsbBus::new(usb_clock, pm, dm, dp, usb))
}
```

Still having some trouble with ADC.

jaredwolff, 2021-02-10:
No dice on the ADC:

```
        // Measurement pin
        let meas_3v3 = pins.meas_3v3.into_alternate();
```

I'm initializing like this and then adding it to a resource in RTIC:

```
struct Resources {
    meas_3v3: hal::Meas3V3,
```

Then,

```
// Get v1 version of pin
let mut meas_3v3: Pb1<_> = cx.resources.meas_3v3.into();
```

Gives me this guy:

```
let mut meas_3v3: Pb1<PfB> = cx.resources.meas_3v3.into();
    |                                                                            ^^^^ the trait `core::convert::From<&mut hal::gpio::v2::Pin<hal::gpio::v2::PB01, hal::gpio::v2::Alternate<hal::gpio::v2::B>>>` is not implemented for `hal::gpio::Pin<hal::gpio::v2::PB01, hal::gpio::v2::Alternate<hal::gpio::v2::B>>`
```

jaredwolff, 2021-02-10:
Converting to and from GPIO versions seems a bit cumbersome. I think your intention is that these interfaces will be updated at some point @bradleyharden?

bradleyharden, 2021-02-11:
@jaredwolff, I'm going to try to look at this in more detail later tonight. But on a quick skim, I'm having a little trouble following what you're trying to do. Could you give me a little more context? What board are you using? Is the code posted anywhere?

jaredwolff, 2021-02-11:
I'm trying to use the ADC on an ATSAMD21J18A. Before this I was using my modified code from the PR. I'll reformat the above to, hopefully, be more useful.

bradleyharden, 2021-02-11:
@jaredwolff, I wasn't reading your error message closely enough, nor was I thinking about the consequences of using RTIC.

The problem here is that you're trying to convert an **&mut** `v2::Pin<PB01, AlternateB>` into an **&mut** `v1::Pin<PB01, AlternateB>` using the `From`/`Into` traits. Those traits are for value-to-value conversion, not reference-to-reference conversion. The trait you need is `AsMut`, which is

> Used to do a cheap mutable-to-mutable reference conversion

In the current code, I only provide implementations of `From` & `Into` between `v1::Pin` and `v2::Pin`. Most interactions with pins are by value, because you need to change the type when changing the mode, which requires ownership.

I should have also implemented `AsRef` and `AsMut` between the two pin types as well. That was an oversight on my part. *However*, the existing #401 PR will handle this for me, purely as a byproduct.

The result is that you will be able to write something list this:

```rust
#[task(resources = [adc, meas_3v3])]
fn measure_voltage(cx: measure_voltage::Resources) {
    let measure_voltage::Resources { adc, meas_3v3 } = cx;
    let res: Result<u16, _> = adc.read(meas_3v3.as_mut());
}
```

I think that ends up being pretty ergonomic. Does that work for you?

@twitchyliquid64, could we get #401 merged so that @jaredwolff can try it out?

jaredwolff, 2021-02-11:
Yea that's not bad. Happy to kick the tires when it's ready. :)

bradleyharden, 2021-02-13:
I'm copying my comments on Matrix here, for posterity.

Looks like it doesn't work without a type annotation:

```rust
pub fn test() {
    use target_device::Peripherals;
    use hal::adc::OneShot;
    let mut peripherals = Peripherals::take().unwrap();
    let mut clocks = clock::GenericClockController::with_internal_32kosc(
        peripherals.GCLK,
        &mut peripherals.PM,
        &mut peripherals.SYSCTRL,
        &mut peripherals.NVMCTRL,
    );
    let mut adc = adc::Adc::adc(peripherals.ADC, &mut peripherals.PM, &mut clocks);
    let pins = gpio::v2::Pins::new(peripherals.PORT);
    let mut pb01 = pins.pb01.into_mode::<gpio::v2::AlternateB>();
    let pb01 = &mut pb01;
    let test: &mut gpio::v1::Pb1<_> = pb01.as_mut();
    let _result: Result<u16, _> = adc.read(test);
}
```

The problem is that `v1` and `v2` pin types can `AsMut` to each other but also to themselves. So the call to `.as_mut()` is ambiguous, and the input to `.read()` is generic. I don't see an easy way around that limitation. The best way is probably to modify the `adc` module to implement `OneShot` for both `v1` and `v2` pins.

But for the moment, this line doesn't seem like a huge burden:

```rust
let test: &mut gpio::v1::Pb1<_> = pb01.as_mut();
```

jaredwolff, 2021-02-13:
> But for the moment, this line doesn't seem like a huge burden:

Yea it's not bad. It just gets confusing when it's "What pin should I use, what works with what? Etc etc" I'm sure that will be alleviated once v1 is completely removed.

jaredwolff, 2021-02-13:
Cool, got it to compile with:

```
// Get v1 version of pin
let meas_3v3: &mut Pb1<_> = cx.resources.meas_3v3.as_mut();
```

Where `cx.resources.meas_3v3` is `hal::Meas3V3`:

```
    #[cfg(feature = "unproven")]
    PB01 {
        name: meas_3v3,
        aliases: { AlternateB: Meas3V3 }
    }
```

bradleyharden, 2021-02-13:
Yeah, when pin numbers get hidden behind aliases, it makes it more difficult. #405 will solve it completely.

PR #405 (2021-02-13): Add ADC OneShot impls for GPIO v2 Pin types
This addresses the issue in #403. Requiring a type annotation is a minor inconvenience, but making this change was also pretty trivial.

Edit: closes #403
closed 2021-02-13 by twitchyliquid64