atsamd-rs/atsamd #414: Unkown USB Device (Set Address Failed)

AdinAck, 2021-03-04:
I have just cloned this repository and have flashed the usb_echo example to my feather_m0 over hf2. The led is blinking just fine (the blinky_basic script worked fine too) but windows does not recognize the feather as a USB device while usb_echo is running. Clearly, USB is working at some point since I was able to flash it with the script, and in bootloader mode, I can see the file system just fine, so I am at a loss as to what is causing this.

I am probably missing something, as I am very new to Rust and embedded systems in general, so this may not be an actual issue with this repository. If that is the case then please let me know and I will close the issue, although I would appreciate a nudge in the right direction :)

twitchyliquid64, 2021-03-11:
What version of rust are you using?

> but windows does not recognize the feather as a USB device while usb_echo is running

What does this look like?

AdinAck, 2021-03-19:
1. My rust version is 1.50.0
2. A notification pops up saying "USB Device Malfunctioned" and in device manager it is called "Unknown USB Device (Set Address Failed".

twitchyliquid64, 2021-03-19:
Hmmm, might have to check things out on `1.50.0`. Can you use rustup and try 1.46 or 1.44 and see if that works?

Did you change anything in the example, like pins or clocks or interrupts?

> A notification pops up saying "USB Device Malfunctioned" and in device manager it is called "Unknown USB Device (Set Address Failed)".

For context, I would expect this if our device did not respond to a configuration packet to set the bus address. This could happen if the device halted due to a panic or was too slow in servicing USB messages.

ianrrees, 2021-03-22:
I've got a couple brand new Feather M0 boards on hand (but my Windows machine is a virtual one), will try to investigate tomorrow.

ianrrees, 2021-03-22:
@AdinAck - interesting find, thanks!  At one level, this is a "works for me" report, however in doing the test I found a strange behaviour that may indicate a problem in atsamd-rs.

My computer runs Ubuntu 20.04, however I have a Windows 10 VM (in QEMU/KVM) which does communicate fine with the usb_echo example.  That's not a great test though; I'm pretty sure the lower-level USB init stuff is handled by Linux in that case rather than Windows.  This evening I can probably borrow a real Windows machine to see how it behaves.

What exact command are you running to build the example firmware?  It might be worth trying with `--release` specified - both debug and release work the same for me, but I've noticed that debug builds are often substantially slower than release ones so perhaps your computer has more strict timing requirements?

I also noticed that cargo-hf2 version 0.2.0 which I had installed didn't work with a couple-months-old Feather M0 Basic, but cargo-hf2 0.3.1 works perfectly.  Since you're successfully flashing things, that's probably not your issue - just putting it here for posterity.

The strange behaviour I'm seeing relates to `lsusb`'s output - I've been working on another USB-connected SAMD21 firmware (A) with Rust firmware, and had noticed that lsusb doesn't show the vendor/product strings in the non-verbose output.  A's VID/PID wouldn't match anything known to lsusb/Linux, so I had just assumed that the displayed strings are baked in to lsusb or some database available to it.  But, when I have a Feather running the usb_echo example (B), suddenly the lsusb entry for A shows the vendor/product strings from B (and B shows "16c0:27dd Van Ooijen Technische Informatica CDC-ACM class devices (modems)").  With `lsusb -v`, I do see the appropriate strings for the idVendor and idProduct fields in both cases.

AdinAck, 2021-03-22:
Ah, adding `--release` seems to have fixed it, can't believe I didn't have that, thank you!
So Windows now recognizes that it is a USB device, but how do I get it to echo?
I connected to it with PuTTY and sent some characters but I don't think it responded, am I using it right?

Thanks!

closed 2021-03-22 by AdinAck

ianrrees, 2021-03-22:
No worries!  I believe that Windows should show a new COM port under Device Manager when the device is connected, then you can fire up PuTTY and select that COM port (baud rate etc doesn't matter, since there's no physical UART involved), typing in the window should show the same characters on the screen as you typed.

Maybe we could improve the example by toggling the LED when characters are entered, or perhaps capitalizing the echoed back characters, so that it is more obvious that the firmware is doing something.

AdinAck, 2021-03-22:
Oh! Cool, then it is working, thank you for the help!