atsamd-rs/atsamd #416: How to get a pin to implement both InputPin and OutputPin?

cbc02009, 2021-03-13:
I'm still fairly new to rust, so I hope this isn't something obvious that I'm missing.

I'm using the samd21_mini crate and I'm trying to read a temperature from either a DHT11 or DS18B20 sensor (using the associated crates), but both use one-wire style interfaces. The example for both sensors have function prototypes that look like this:

```
fn find_devices<P, E>(
    delay: &mut impl DelayUs<u16>,
    tx: &mut impl Write,
    one_wire_pin: P,
)
    where
        P: OutputPin<Error=E> + InputPin<Error=E>,
        E: Debug
{
```
 
It seems like no matter what `into` I choose for the pin I'm using it only implements one or the other. Is this a hardware-level issue with the samd controllers that can't be worked around in code, or is there something obvious I'm missing? I didn't see anything in the d21 datasheet about using a pin in both input and output mode simultaneously, so I'm leaning towards it being a hardware issue, but I figured I'd check. 

My other thought is it might have something to do with v1 gpio vs v2, but I don't really know anything about that. I think the atsamd crates use v1 for the boards? It seems like the examples use v2, but I'm not sure how I would convert the board crate to v2 to test it out.

olback, 2021-03-21:
I ran into this issue as well today. Was stuck for a few hours and tried enabling the `unproven` feature and using the `into_readable_open_drain_output` method and it works just fine. Would love to see better documentation but I'm not sure where this would be documented.

bradleyharden, 2021-03-21:
@olback, I think this comes from the fact that our version of `embedded_hal` is out of date. I'm fairly certain the `InputPin` trait is no longer `unproven`.

Also, when dealing with the GPIO pins, you might consider trying out the new, `v2` API. It fixes a few issues with the `v1` API and hopefully provides a cleaner, simpler interface. The docs for `master` can be found [here](https://atsamd-rs.github.io/docs/), and the GPIO documentation can be found, e.g. [here](https://atsamd-rs.github.io/docs/samd21g/thumbv6m-none-eabi/doc/atsamd_hal/common/gpio/index.html). The `ReadableOutput` mode implements both `InputPin` and `OutputPin` in `gpio::v2`.

cbc02009, 2021-03-21:
> @olback, I think this comes from the fact that our version of `embedded_hal` is out of date. I'm fairly certain the `InputPin` trait is no longer `unproven`.
>
> Also, when dealing with the GPIO pins, [...]

Thanks for the suggestion! I'm using the samd21_mini crate for UART and SPI, and from my (limited) understanding, the crate uses the v1 API? Is there a guide to modifying it for use with the v2 API?

> 
> 
> I ran into this issue as well today. Was stuck for a few hours and tried enabling the `unproven` feature and using the `into_readable_open_drain_output` method and it works just fine. Would love to see better documentation but I'm not sure where this would be documented.

Thanks! I'll give this a try if I can't get the v2 API working.

bradleyharden, 2021-03-21:
@cbc02009, just to clarify, the v2 API won't help for your issue. The InputPin trait is marked unproven in the upstream embedded_hal crate (in the version we're using). We just haven't updated it in a while. You'll need to enable unproven using either v1 or v2.

Also, if you're new to atsamd-rs and/or Rust and you're using an existing BSP, it will probably be easier to stay on v1. I think v2 is where we're headed, but there aren't many examples of using it yet.

cbc02009, 2021-03-21:
> @cbc02009, just to clarify, the v2 API won't help for your issue. The InputPin trait is marked unproven in the upstream embedded_hal crate (in the version we're using). We just haven't updated it in [...]

Thanks for the advice. I have a decent amount of embedded c experience (especially with SAM microcontrollers) so I thought this might be an easier way for me to learn rust than with linux hello-world style stuff. I'll give the unproven feature a shot.

closed 2021-03-21 by cbc02009