atsamd-rs/atsamd #426: API cleanup

vccggorski, 2021-04-07:
Effort to standardize API across HAL & BSCs. 

- Remove chip-specific modules
  PR #424
- Flatten HAL by removing redundant `common` module
  PR #425
- Update the BSPs to re-export the HAL itself, rather than modules within the HAL
  PR #357
- Make the `thumb` modules private, to indicate that they are an implementation detail; consider the remaining modules public and let them become the official API
  PR XXX
- Rename imports across the HAL and the BSPs to increase consistency and clarity. (open discussion)
  PR XXX

vccggorski, 2021-04-07:
As far as I could see, discussion sorta started already in #357. Nonetheless, properly tracked issue seems more appropriate for such thing.

bradleyharden, 2021-04-20:
Here's another proposed change. I would like to change `target_device` to `pac`. At the very least, maybe we could support both. Thoughts?

Sympatron, 2021-04-20:
Maybe we could review the `unproven` feature. Last time I looked a lot of basic stuff was "unproven" (I think because the traits were unproven in `embedded-hal`).

Sympatron, 2021-04-20:
> Here's another proposed change. I would like to change `target_device` to `pac`. At the very least, maybe we could support both. Thoughts?

Seems like this is already part of PR #357.

bradleyharden, 2021-04-20:
I meant at the HAL level, not the BSP level. I don't think that PR affects the HAL name.

bradleyharden, 2021-04-21:
@Sympatron, I would expect unproven to disappear with embedded-hal 1.0, but who knows when that will be.

Sympatron, 2021-06-24:
Thinking about the `unproven` thing more, I came to the conclusion that we probably should not mark whole modules as "unproven" just because they implement some unproven trait. Everything else in these modules is "proven" (to the same degree as most of the hal). So I would vote to mark only the e-h trait impls as `unproven`.

TDHolmes, 2021-09-26:
#491

TDHolmes, 2021-10-09:
> Rename imports across the HAL and the BSPs to increase consistency and clarity. (open discussion)
PR XXX

done in #491

ianrrees, 2024-08-27:
It seems like the work discussed here has all been done, any reason not to close this?

glaeqen, 2024-09-01:
I think this issue got outdated considering that I stopped following up on the changes that happened in the meantime. We can close it.

closed 2024-09-01 by glaeqen