atsamd-rs/atsamd #429: RFC: New clocking API

vccggorski, 2021-04-16:
- Feature Name: new_clocking
- Start Date: 2021-04-16
- Authors: @vccggorski, @vcchtjader
- RFC PR: [atsamd-rs/atsamd#XXXX](https://github.com/atsamd-rs/atsamd/pull/XXXX)
# Summary
[summary]: #summary

New clocking API allowing to build a typesafe chain of dependent clock sources, generic clock generators and peripheral channels.

# Motivation
[motivation]: #motivation

Current clocking API assumes the following workflow:

```rust
let mut clocks = GenericClockController::with_{external,internal}_32kosc(...PAC...);
let gclk9 = clocks.configure_gclk_divider_and_source(ClockGenId::GCLK9, 1, ClockSource::DFLL48M, false);
let adc0_clock: Adc0Clock = clocks.adc0(&gclk9);
```

This API has the following limitations.

1. By default the `GenericClockController` constructor configures and provides an _opinionated_ set of clock sources and generic clock generators that cannot be changed afterwards. Following clock configuration is set for `thumbv7em`.
    - `GCLK0` (Generic Clock Generator 0) is driven by `DFLL48M` (48MHz) through `GCLK5` (divider: 24 -> 2MHz) through `PCh1` (Peripheral Channel 1) through `DPLL0` (multiplier: 60 -> 120MHz)
    - `GCLK1` is driven either by `XOSC32K` (use\_external\_32kosc) or `OSCULP32K` (use\_internal\_32kosc)
2. It is not possible to set up additional clock sources within the framework of current clocking API (without _hacking around PACs_).
3. Once you setup the `Clock Source - GCLK` pair, it is impossible to change it later.
4. Once you setup the `GCLK - PeripheralChannel` pair, it is impossible to change it later.

Main clock locked to 120MHz by HAL, even though being acceptable in basic scenarios, is a serious design flaw that severely diminishes flexibility of a HAL and might either encourage users to hack unsafe solutions on top of existing abstractions or discourage them from using the HAL altogether.

Having these points in mind and also the fact that current clocking API implementation would benefit from refactoring anyway, striving for improvement seems to be fully motivated and justified.

# Detailed design
[design]: #detailed-design

## Introduction

ATSAMD clocking system assumes 3 types of major building blocks:
- Clock sources (referred to further on by `ClkSrc`)
- Generic Clock Generators (referred to further on by `GClk`)
- Peripheral Channels (referred to further on by `PCh`)

Properties of ATSAMD clocking system:
- `GClk`s depend on `ClkSrc`s in a `N:1` relationship
- `PCh`s depend on `GClk`s in a `M:1` relationship
- **Some** `ClkSrc`s can depend on **some** `PCh`s
  Specifically:
    - For `thumbv7em`-based MCUs
        - `PCh:PCh0` can serve as a reference clock provider for `ClkSrc:DFLL48M`
        - `PCh:PCh1` can serve as a reference clock provider for `ClkSrc:DPLL0`
        - `PCh:PCh2` can serve as a reference clock provider for `ClkSrc:DPLL1`
        - `PCh:PCh3`? Take a look at [#unresolved-questions][unresolved]
    - For `thumbv6m`-based MCUs
        - `PCh:PCh0` (on `thumbv6m` Peripheral Channels are called GCLK Multiplexers) can serve as a reference for `ClkSrc:DFLL48M`
- **Some** `ClkSrc` can depend on **some** other `ClkSrc`s
  Specifically
    - For `thumbv7em`-based MCUs
        - 32Khz signal of `ClkSrc:XOSC32K` can serve as a clock source for `ClkSrc:DPLL{0, 1}`
        - `ClkSrc:XOSC{0, 1}` can serve as a clock source for `ClkSrc:DPLL{0, 1}`
        
## Mapping HW model to Rust

These properties are going to be reflected in Rust by traits (representing abstract concepts of `ClkSrc`, `GClk` and `PCh`) and structs implementing these traits (representing concrete HW instances of these concepts like `ClkSrc:XOsc0`, `GClk:GClk0` or `PCh:PCh0`).
```rust
trait ClkSrc {}
trait GClk {}
trait PCh {}
// Eg.
struct XOsc1 { /* ... */ }
[... 175 lines cut ...]

What other designs have been considered? What is the impact of not doing this?

Retaining the similar but subtly different clocking abstractions for `thumbv7em` and `thumbv6m`, with the drawbacks as discussed above.

# Unresolved questions
[unresolved]: #unresolved-questions

What parts of the design are still TBD?

NOTE: These questions are mostly related to `thumbv7em`.
- `genctrl::SRC_A`: PAC defined clock sources, what are `GCLKIN` and `GCLKGEN1` (in a diagram on a [page 142][v7datasheet] `GCLK1` seems to be able to become an input for other `GCLK`s even though it is not apparent on a diagram from [page 136][v7datasheet])?
- How to incorporate CLK\_RTC\_OSC (`RTC` clock source), CLK\_WDT\_OSC (`WDT` clock source) and CLK\_ULP32K (`EIC` clock source)?
- What's the nature of relationship between `USB` peripheral and `DFLL48M`. Can it be incorporated to the typesafe hierachy in a coherent manner ([page 136][v7datasheet])?
- What is `PCh3` specifically used for (what is the GCLK\_DPLLn\_32K signal)?
  Hypothesis:
  - `DPLL0` can rely either on a reference clock from `PCh1` (GCLK\_DPLL0) or `PCh3` (GCLK\_DPLL0\_32K)
  - `DPLL1` can rely either on a reference clock from `PCh2` (GCLK\_DPLL1) or `PCh3` (GCLK\_DPLL1\_32K)

# References
If there's a reference to an external document with a specified page, it is implicit reference to [`SAM D5x/E5x Family Data Sheet` (`DS60001507F`)][v7datasheet] in a context of `thumbv7em` based MCUs.
For `thumbv6m`-based MCU, it is a reference to [`SAM D21/DA1 Family Data Sheet` (`DS40001882F`)][v6datasheet].

[v6datasheet]: https://ww1.microchip.com/downloads/en/DeviceDoc/SAM_D21_DA1_Family_DataSheet_DS40001882F.pdf
[v7datasheet]: http://ww1.microchip.com/downloads/en/DeviceDoc/SAM_D5xE5x_Family_Data_Sheet_DS60001507F.pdf

bradleyharden, 2021-04-17:
@vccggorski and @vcchtjader, as discussed on Matrix, I already have a nearly-complete `clock::v2` module. I haven't worked on it in a few weeks, so I forgot that it is only for `thumbv7em` right now. I haven't read through the D21 or D11 datasheets, though, so I don't know how easy it would be to generalize it.

Before digging in, I would recommend you understand what I've started to call the `AnyKind` trait pattern. I use it in `gpio::v2` and `sercom::v2`, and @jbeaurivage has also used it in `dmac`. It ends up being *really* useful for this kind of type-level programming. I think my best explanation of it is [here](https://atsamd-rs.github.io/docs/samd51n/thumbv7em-none-eabihf/doc/atsamd_hal/common/gpio/v2/pin/trait.AnyPin.html), the documentation for `AnyPin`. I'm working on adding some documentation to the `typelevel` module to explain the general concept. Then, each implementation can point to that centralized documentation.

I've read your proposal in a little bit more detail, and I have to say that you hit the nail on the head in every count.
- I have a `clock::Source` wrapper struct for several different types of clock source, with an `AnySource` trait; a `Gclk` type and an `AnyGclk` trait; and a `Pclk` type and an `AnyPclk` trait.
- I use some type-level magic and the `typenum` crate to implement a compile-time `LockCount` trait that handles the M:1 problem. Once a `Source` or a `Gclk` has been locked, it can't be modified. But it can be locked and unlocked multiple times, all tracked at compile time.
- I also handle most of the cases where `Pclk`s wrap back around and are used by clock `Source`s. However, I think I missed that `Dfll` can take `Pclk0` as a source. I didn't pay much attention to `Dfll`, honestly. It's not a huge priority for me personally.

For the `MCLK` stuff, I have `ApbClk` and `AhbClk` types that represent enabled APB and AHB clocks, while `ApbToken` and `AhbToken` types represent the disabled clocks.

I tried to add some comments just now. I hope the are helpful. After a few minutes, I realized that I could spend hours writing comments for everything, so I decided to stop for now. Honestly, that's why I haven't worked on it in a while. There's just so much documentation to write. But I think that documentation is really important if we want this to be accessible to people.

Have a look and let me know what you think. Maybe we can collaborate and use this as a starting point. @jbeaurivage and I collaborated quite a bit on the `dmac` module.

bradleyharden, 2021-04-17:
I forgot to include the [link](https://github.com/bradleyharden/atsamd/tree/clock_rebase).

Also, [here](https://github.com/bradleyharden/atsamd/blob/b0f416b859bd5e003d4bbe4e505da67d6b2b8a66/hal/src/common/thumbv7em/clock.rs#L7) is a good example of the entire library in action. It's extensively commented, so I think it should be easy to follow. This is an entirely real use case. It's almost exactly my actual clocking structure for my work project.

bradleyharden, 2022-04-24:
@vccggorski, we missed celebrating the one year anniversary of this PR! 😢

We'll get there eventually. 🥳

vccggorski, 2022-04-24:
You're right, damn :/ I have to focus and allocate some time for it, argh.

bradleyharden, 2022-12-26:
Closed with #450

closed 2022-12-26 by bradleyharden