atsamd-rs/atsamd #430: Change mainline branch name from master to main

TDHolmes, 2021-04-17:
Should be pretty easy to do for anyone with admin privileges

ianrrees, 2022-03-28:
I agree this would be a nice change to make.

We do have quite a few URLs that refer to "master" in the repo, so those will need to be updated as well.  Maybe the thing to do is make a PR for the URL changes and have an admin switch the default branch when that's merged.

TDHolmes, 2022-03-29:
@bradleyharden had some thoughts on this back when I initially proposed that we never talked about - Bradley do you want to discuss on matrix?

bradleyharden, 2022-04-16:
Sorry it has taken me so long to respond. It's always a combination of forgetting about this topic and being too busy to work on a response.

I thought about writing this in essay form, but I think a bullet-point approach might make it more organized? I'm not sure.

Either way, I'll just jump right in.

- Assumptions & perspective
    - I live in the US and approach this from a **very** US perspective.
    - I assume this request is based on some desire to be "anti-racist" and remove what is seen as racist language from Git.
        - Maybe that's not the case? But if not, then what's the point?
    - I generally perceive this movement as being rooted in the US, as a means to grapple with our legacy of slavery.
    - For whatever its worth, I am an upper middle class white man with a well paying job in engineering.

- First of all, the association of `master` branch with "slave master" is dubious at best.
    - If you know enough about the history of Git, you can indeed trace it back to BitKeeper, which I believe had `master` and `slave` branches.
    - Git inherited `master` as the primary branch name, but that doesn't necessarily mean it was intended as one-half of `master`/`slave`.
        - In fact, the non-existence of `slave` branches would imply the opposite.
        - Has anyone actually asked Linus why he chose `master`?
    - Does anyone new to Git *actually* associate the `master` branch with "slave master"?
        - When I first learned Git, I certainly didn't. I assumed it was the "master copy" branch.

- Second, I find this movement somewhat US-centric.
    - Are people in other countries concerned about this or other instances of specific language?
    - If so, are we all prepared to change our language to accommodate them?
    - For instance, I've heard the term "collaborator" has a strong negative connotation in Poland, because of Nazi collaborators.
        - Should GitHub replace its language around "collaborators"?

- Third, I find it interesting that the two posts so far don't actually provide a motivation for the change.
    - I think that speaks to two issues:
        - One, it's assumed without discussion that this is the "right" thing to do.
        - Two, people are often afraid to address the issue openly and head on.
    - Both of these feed into my next point.

- I consider the `master` to `main` movement, and movements like it, a distraction from the real problem in the US.
    - The US is currently reckoning with its legacy of slavery, but I find people often resort to empty, token gestures of "anti-racism" rather than deal with the far more difficult and deep-rooted problems.
    - I don't think people are asking the obvious yet difficult questions we should be asking:
        - Why do black and white Americans living in the **same city** have completely different English dialects and accents?
        - Why do they have different cultural norms?
        - Why do they live in different neighborhoods and go to different schools?
    - The answer, and the root of **all** these problems, is that we live in a segregated society.
        - Racism, both explicit and implicit, is a byproduct of segregation.
    - What exactly would renaming `master` to `main` accomplish?
        - It certainly won't reduce segregation or meaningfully improve the circumstances of black America.
        - The absolute best case scenario is that it would remove a tiny paper cut for a small portion of black ECE/CS students who look deep enough into Git's history. And even then, the "problem" is dubious at best (see above).

- I consider any action that doesn't directly address segregation a waste of time.
    - The NY Times had a great piece on a [Georgia woman](https://www.nytimes.com/2021/07/04/dining/georgia-farm-slaves.html). It was one of the few times I read about someone who I thought really *got it*. I liked the quote from her husband:
        - > “It’s like, ‘I’m not going to change my life, but tell me a dollar amount that would absolve me of guilt,’” he said. “That kind of transaction, whether it’s about the environment or racial inequality, is not going to create change.”
    - The lesson is this: **We can't transform our society without also transforming our own lives.**

- When evaluating a life decision, everyone already asks "How does this affect me and my family". But I contend we should also be asking "How does this affect everyone else?"
    - I often hear people discuss moving to a new neighborhood for its "good" schools.
        - This attitude is precisely the problem.
        - On its face, it looks like "providing the best opportunity for your children". But the reality is that it also removes opportunity from other children.
        - As a consequence of this mindset, the most well educated and economically successful Americans end up walling themselves into wealthy communities.
        - Our neighborhoods and schools become stratified along lines of economic circumstance, which inevitably correlates with race.
         - It's no surprise, then, that these schools under-perform.

- From this line of reasoning, I conclude we must all consider the broader implications of our choices and challenge ourselves to be agents of integration.
    - Instead of moving to *find* a "good" school, work to *create* a good school for your own children and everyone else's.
    - Instead of buying a house in a "good" neighborhood with an HOA to "protect" your investment, buy a duplex in a mixed income neighborhood and create affordable housing for struggling families.
    - Instead of defaulting to a comfortable existence surrounded by people with similar means and culture to your own, put yourself in situations where you are uncomfortable and different.

If we don't do these things, I doesn't matter how many branches we rename, we will never create meaningful change.

ianrrees, 2022-04-29:
Thanks for the thoughts @bradleyharden !

My perspective is similar - I'm a white male from the US, Georgia specifically, with a comfortable tech job, though early in my tech career I was the sysadmin for a bunch of sociologists in academia, and I immigrated to Aotearoa/New Zealand about a decade ago.

One of the main voluntary things that keeps me busy is pest trapping; we've got a lot of invasive mammals which are eating what's left of our native ecosystems.  There's a movement/goal "Predator Free 2050" which is what it sounds like, and a lot of optimistic ecologists think it's pretty much impossible.  Most of the traps I maintain are in a public park and often I encounter folks who are out walking, sometimes they've got something to say about the trapping.  For the most part, people are grateful and/or curious, but some also see it as misguided/suboptimal/futile.  The second group's intent is not malicious, but their suggestions carry some implications that are unhelpful and that could be disrespectful; I think of these as "can't you just" suggestions (danluu has a fairly popular [post on this](https://danluu.com/cocktail-ideas/)).  Basically, the problem is that I actually have a rational and clear idea of why and what I'm doing, including the context around it, but those casual "can't you just" suggestions are some combination of misinformed, morally questionable, and missing the point.

All that's just to say that "everything is a thing", and I wouldn't want to weigh in about the right course of action for an issue that I'm not particularly informed about nor affected by.  The change we're talking about here is an easy one to make, so for me it's enough to understand that some people feel it's worthwhile.

A few bullet points:
- At this stage, my muscle memory is retrained for "main".
- I think it's pretty safe to say that the industry practice is moving from "master" to "main", so a decision /not/ to change is the more activist stance.
- The "master copy" interpretation isn't really a good fit for branching in git either, because the master copy is the thing that gets replicated identically, but a branch evolves over time.