atsamd-rs/atsamd #433: Establishing a BSP tier system

bradleyharden, 2021-04-25:
### Introduction

This repository currently houses 28 different BSPs, all of which compile with the current version of the HAL. While this is one of the great features of `atsamd-rs`, supporting that much downstream code is becoming a burden. Specifically, it has become difficult to make any breaking changes to the HAL, because every single BSP would need to be updated along with it.

We've discussed this problem a number of times, both in issues and on the Matrix channel. It seems like consensus is gathering around a tiered system of support for BSPs. In this system, Tier 1 BSPs would be guaranteed to compile on `master`, while Tier 2 BSPs would be tied to a specific, working release of the HAL and updated only as desired.

Naturally, there are a few questions:

1. Is this solution acceptable to everyone? Do we have a consensus?
2. How many Tier 1 boards should we support?
3. How should we decide which boards are considered Tier 1?

I will respond with my own answers, and I encourage others to do the same.

1. I think this is the best solution we've found so far. I don't think it's feasible to support 28 BSPs indefinitely. Ultimately, that will stifle progress in the HAL. Moreover, it seems like many of the existing BSPs are little more than skeletons. 15 of the 28 BSPs have 3 or fewer examples. Eight of them only have one example.
2. Personally, I think we should limit it to approximately 5. I think that will keep things manageable. There may be Tier 2 BSPs with great support, but that support would come separately from changes to `master`.
3. A few qualifications and goals have been brought up so far:

- Choose existing BSPs with broad support for many different HAL APIs
- Choose BSPs to get good coverage of different chip variants
- Choose popular boards

Here is some data to base the decision on.

### Data

There are six boards with 10 or more examples. It looks like five of them are D51 and only one is D21.

edgebadge (D51)
feather_m0 (D21)
feather_m4 (D51)
metro_m4 (D51)
pygamer (D51)
wio_terminal (D51)

Here are the top 10 most popular boards (according to crates.io):

1. metro_m0
2. metro_m4
3. gemma_m0
4. arduino_mkrzero
5. itsybitsy_m0
6. trellis_m4
7. samd21_mini
8. feather_m0
9. circuit_playground_express
10. sodaq_one

It doesn't look like there is that much overlap between popular boards and boards with a lot of examples. Perhaps some of the BSPs have good support but few examples? Or perhaps the people downloading the most popular boards don't need much functionality?

I think I would be most likely to choose from the six boards with a lot of examples. Although I might also want to drop one or two of the D51 boards and add a second D21 board.

Thoughts?

Paging notable contributors. Sorry if I miss anyone.
@sajattack, @jessebraham, @twitchyliquid64, @Sympatron, @jacobrosenthal, @jbeaurivage, @TDHolmes, @jaredwolff

sajattack, 2021-04-26:
I think we should go with the most complete existing BSPs, and the BSPs current contributors have boards for. Popularity is a small factor imo.

jaredwolff, 2021-04-26:
Thanks @bradleyharden for thinking about this.

As a comparison, one thing that I've found useful in Zephyr (where I spend most of my embedded time.. so looking forward to doing more in Rust though!) is that instead of having a similar example per BSP, they make it generic enough so you can essentially add the BSP to the example. As long as it has the supported peripherals and aliases "it just works".

For instance it's easy to implement a blinky example if all BSPs support an LED0 (or similar) alias. A UART sample could be done similarly. USB serial would be another (and so on). Since many of the BSPs have initialization functions for such peripherals, technically if they're all named the same way, you could use the same example for many boards.

Heres an example of [Blinky](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/basic/blinky) that can be used on any board that has an `led0` alias.

Of course, I haven't seen all of the conversations around this to please ignore if this idea has been beaten to death. 

I otherwise agree that reducing the set of BSPs supported makes good sense. I would add that it also makes sense that whatever is Tier 1 is used by the main contributors. (Agreeing with @sajattack here) This is especially true if the examples are unique to the boards in question (whereas my earlier approach may not work).

jbeaurivage, 2021-04-26:
I wholeheartedly agree. If @jaredwolff's suggestion is doable, that would be ideal. Unless I'm missing something, apart from the examples, piling on more BSPs (of the same chip type) isn't beneficial if the only added value is predefined pin aliases. We're probably better off thoroughly documenting the bsp_pins! macro and how to make a DIY BSP.

Otherwise, apart Tier 1 BSPs, I think we might eventually want to consider handing off Tier 2 BSPs to downstream community projects - which probably have better incentive to maintain a quality BSP. A good start might be to split them off into their own git repo?

As far as which existing BSPs should be made Tier 1, I'd like to see at least 1 SAMD11 board in there if we have any? If support for new chipsets eventually gets added, we'll have to support them too.

bradleyharden, 2021-04-26:
@jaredwolff, interesting idea about unifying the examples. I like it. `embedded-hal` and our own generic code should already make that possible for most tasks. And I think the `bsp_pins!` aliases can probably cover differences in pinout.

@jbeaurivage, looks like the only SAMD11 board right now is the samd11_bare.

Sympatron, 2021-04-26:
I think unifying the examples is a great idea. After all some of the examples are relevant for any BSP or at least most of them, like blinky, usb, rtic, timer, pwm, serial, adc just to name a few. These examples should essentially contain the same code for most BSPs so getting rid of duplicates would simplify maintaining them a lot. This will probably also help users to find them, because you would currently have to look into all BSPs for examples if they are not present in the BSP you are using.

bradleyharden, 2021-04-26:
How can we test that they work with all of the Tier 1 BSPs? If we could guarantee Linux, we could just add symlinks in the examples folder. But I don't see an easy way to do that otherwise. Any ideas?

vcchtjader, 2021-04-26:
Disclaimer, I don't use Windows but it seems there's some support for symlinks, although it seems cumbersome in general as discussed [here](https://stackoverflow.com/a/59761201). 

An alternative could be an "examples folder" where users without symlink support can grab a copy of examples interesting them, while symlinks points to that same folder.

Splitting out Tier 2 examples into its own repo as suggested by @jbeaurivage sounds good.

bradleyharden, 2021-04-26:
> Be careful, support for Symlinks in git on Windows is relatively new. There are some bugs that still affect some git clients. Notably, symlinks with relative (`..`) paths are mangled in some programs

Sounds fun...

LongLiveCHIEF, 2021-04-26:
Examples should be common wherever possible as a rule, and then assumed that any example in the BSP's are unique to that BSP.

bradleyharden, 2021-04-26:
Would anyone be interested in volunteering to figure out how to combine the examples and get them to run with cargo on all OSs?

rtsuk, 2021-04-26:
Could it be as easy as the way [embedded-graphics](https://github.com/embedded-graphics/examples/blob/main/eg-0.6) with a separate crate but one that has features to pull in the desired BSP?

bradleyharden, 2021-04-26:
Probably. And we could select the BSP with feature flags, just like we do with PACs in the HAL.

LongLiveCHIEF, 2021-04-26:
Can you use `[[examples]]` in Cargo workspaces?  If so, doing what `probe-rs` does could really be helpful for using those cargo patching when testing. 

https://github.com/probe-rs/probe-rs/blob/08d9fa2a1ae42d7c2ac230bf4e6c8c8adadeecf4/Cargo.toml#L1-L11

rtsuk, 2021-04-26:
From my reading of https://doc.rust-lang.org/cargo/reference/workspaces.html, not in a virtual manifest like probe-rs uses.

rtsuk, 2021-04-26:
If one put a virtual manifest at the top of this repository, with all the existing packages and the new example package as members, I think it could work very well.

vccggorski, 2021-04-27:
Have you considered *just* putting all of them in a separate repository and having them locked down to latest stable HAL release?

According to my understanding BSCs were supposed to provide simplified, easier to understand (but usually limiting in many regards) API, very often narrowed down to specific board so eg. GPIO names correspond to labels visible on a board etc. They are mostly (AFAIU) targeting students and people exploring embedded systems that want to get something remotely quickly up and running while not being instantly overwhelmed by complexity of an actual HAL. These people probably won't be that interested in bleeding-edge APIs and abstractions as they are just learning basics while creating some kind of simple project. For HAL maintainers, getting rid of such burden would be a great relief allowing them to spend more time on building a better HAL. Which, IMHO, in a long term would render a significant portion of BSCs redundant as HAL ergonomics improve.

I value HAL and its development higher than BSCs because it's a selling point (for a lack of a better expression) of Rust Embedded for Microchip MCUs in a industry. But I fully understand that my view of what should be prioritised might and probably will differ from others'.

rtsuk, 2021-04-27:
> Have you considered _just_ putting all of them in a separate repository and having them locked down to latest stable HAL release?

In what I proposed, all the examples will be locked down to a particular HAL and BSC, selected by a feature flag of the example crate.

I don't see an advantage in having them in a separate repository, though. What is appealing about that to you?

rtsuk, 2021-04-27:
> Would anyone be interested in volunteering to figure out how to combine the examples and get them to run with cargo on all OSs?

I'll taking a shot at this now.

vccggorski, 2021-04-28:
> I don't see an advantage in having them in a separate repository, though. What is appealing about that to you?

The most remotly technical argument I can see would be discouragement from making changes both in HAL and BSCs at the same time and coupling them. Rest is pretty ephemeral. IMO HAL and BSCs are separate constructs so it'd make sense for me to keep them separated.
Similar thing can be said about PACs, although these rarely change and are mostly autogenerated. Splitting these would probably be an overkill, I dunno.

rtsuk, 2021-04-28:
For discussion: https://github.com/rtsuk/atsamd/tree/examples/samples.

cbc02009, 2021-05-02:
I'd like to suggest a new format for the BSPs in order to make them easier to prototype and do simple projects with. I think this fits here, but I can create a separate issue if necessary.

I think it makes sense for BSPs to be a device tree-like `struct` containing the pre-configured defaults for a particular board. I'm working on something similar for the adafruit nRF52840 feather [here](https://github.com/cbc02009/adafruit-nrf52840-express-rs), and I think this format provides two main benefits: 

1. It allows for better code completion, which allows for easier prototyping and getting started, as it's much easier for IDEs to parse `struct` fields than expanded macros.
2. It allows for complete board setup using a simple `let mut board = BSP::Board.take().unwrap()`. The underlying code for setting up peripherals etc, is all still available in the `take()` function, so if someone needs to configure a board differently, the examples still exist for them to do so.

The main disadvantage of this approach is that the BSPs themselves won't be able to be customized (unless there's some way to do so with parameters or features, I'm still very new to rust). However, considering the main use case of boards like the ones supported by these BSPs are early prototyping and simple home projects, I think this approach fits perfectly.

quentinmit, 2021-05-15:
I think part of the reason there's such a wide variation in quality of BSPs is that right now each BSP does things in its own way and there aren't very many conventions. So e.g. it would be very hard to write a single example that uses UART0 on all the boards that have a UART, because there's probably 5+ different APIs in the various BSPs. I think part of the goal here should be to push as much upstream into the HAL as possible, so the BSPs can be simple and consistent. Then you could more plausibly write a simple UART example that runs on all the boards. That would also help reduce the maintenance burden while simultaneously increasing the quality of the skeleton BSPs.

That said, I think this should probably wait a bit while the `v2` APIs are still stabilizing. I envision a future where there's a macro ala `bsp_pins!` that handles 90% of the work of most of the BSPs. (Think a `bsp_board!` macro that takes pins and some simplified description of the onboard peripherals, and pregenerates UART/SPI/I2C/USB convenience functions and types.)

bradleyharden, 2021-08-08:
So I've been thinking about how to proceed with deprecation and migration from `v1` to `v2` modules. As I see it, there are four steps:

1. `v1` is the default. `v2` is available.
2. `v1` is the default, but deprecated. `v2` is available and the encouraged option.
3. `v2` is now the default. `v1` is still available, but deprecated.
4. `v2` is the default. `v1` has been removed.

We are currently at step 1 for `gpio`, and `sercom::{spi, uart}`. The question is how to proceed through the remaining steps.

I don't think moving to step 2 would require a major version bump (minor version in our case, since we're < 1.0). Steps 3 and 4 would both require major version bumps. I don't know that step 3 is all that useful. Because `v2` is the default, existing code will break. Granted, the breakage is fairly small. It only requires changes of the form

```diff
- use atsamd_hal::gpio;
+ use atsamd_hal::gpio::v1 as gpio;
```

But there is still breakage.

Here is one proposed course of action:

- Release 0.13 at step 2. We would have to add deprecation notices to the corresponding modules. This would spam BSP users with warnings, especially for `gpio`, but it would at least make it clear to them that breakage is imminent. I think it would be fairly straightforward to turn off those warnings if needed.
- For any BSP that hasn't yet been migrated to `v2`, update `Cargo.toml` to stay on 0.13.
- Any BSP that has been migrated automatically qualifies as Tier 1. Other boards will no longer be tested on `master` until they are migrated.
- I will try to migrate a few more boards myself. Once we figure out pygamer, I can port edgebadge too. It's nearly identical to pygamer. I also want to port wio_terminal. Are there any other important ones?

jbeaurivage, 2021-08-09:
@bradleyharden, this seems pretty reasonable. Right now, #462 is at step 3. I can of course update it to step 2, but since moving to v0.13 allows for breaking changes, we can just as well remain at step 3. I don't really have a preference either way.

As for the BSPs, I don't know if I like promoting any board that is fully updated to Tier 1. As boards are updated, we might end up in a situation where we effectively have 50+ Tier 1 BSPs again. To me that's not super maintainable, although it's more manageable than what we have now. 

I prefer picking a small set of boards that we commit to keep updated at all times, and not allow for so much mobility between tiers for a board. It's a lot of back and forth, especially if we hypothetically introduce a `v3` module, or if a new module gets a `v2` upgrade. We're back at square 1 if at some point in time all boards are back up at Tier 1 status and a new module upgrade gets introduced. 

I also maintain that it makes sense to split off Tier 2 BSPs into their own git repo, though that's up for debate. To me that's more in line with their purpose of being "community maintained examples to quickly get up and running with a certain board using the HAL", rather than tools for CI, which is more Tier 1 boards' purpose.

bradleyharden, 2021-08-11:
@rtsuk, I took a look at your example repo. I like it as a concept, and it works well for something like `blinky`, but I see a few problems with the approach for more complicated examples.

The problem is that the pin/clock/peripheral used for a particular example can/must change from board to board. The `define_pins` and `bsp_pins` macros can solve this problem for pins, but the problem still remains for peripherals and clocks.

The `blinky` example only uses pins, so it's pretty simple. But something like a UART example would need a way to specify which SERCOM to use. That can be done with macros as well, but we would need to come up with a standardized way to do that. The same applies to the peripheral channel clocks. You need to provide the matching clock for each peripheral, so you would need a macro approach to make itwork over different boards.

Finally, some of the peripheral modules are structured like `Uart`, where they are generic over the PAC `UART` type. But others, like the `pwm` module, have unique `Pwm#` structs for each corresponding PAC `TC`. The compiler can infer the correct `UART` type with the former approach, but the example would have to specify the correct `Pwm#` type in the latter case. We would need some way to paper over this difference. It could probably be done with a type alias, or maybe another macro.

In any event, I think there are too many questions left open to pursue this approach for v0.13. Maybe we can try to revisit it again in the future. This will also be affected by the upcoming `v2::clock` module.

bradleyharden, 2021-08-11:
@jbeaurivage, I think I'd like to avoid such a big step in the move to v0.13.

It's been ages since we had a release, and there have been a lot of fixes, updates and features added that are independent of the `v1`/`v2` issue. Right now, all of the BSPs still compile on `master`, so I would like to release all of those updates without breaking existing code. However, I would also like to indicate that this will be the last release that is compatible with `v1`. I think that points to releasing v0.13 at step 2 from above.

After that, I think we can move quickly with a follow-up release, v0.14, that moves to step 3 or step 4. I would favor skipping step 3, though. By keeping `v1` around, even if it's not the default, we still have many of the problems we are trying to solve. We still have to maintain it. We still have to support it when people ask questions. We still have to provide fixes for incompatibilities like #464. And it still creates confusion, because there are two ways to do the same thing.

I like your reasoning on the BSP tiers. How does this sound for a complete list of tier 1 boards?
- feather_m0
- metro_m0
- pygamer
- wio_terminal

Is that enough? Does anyone want to nominate others?

However, I'm not sure about splitting out the remaining BSPs to their own repo. I see two problems there.
1. We would move the repo location for those BSPs, which would break links in docs.rs for old versions. I guess that's not a huge deal, but it's something to consider.
2. It creates a substantial barrier to moving boards between tiers. I guess you're probably against moving boards between tiers very often, but I could see it happening occasionally. For instance, if we lose all contributors with access to a particular tier 1 board, we may need to demote it. Or we might get a new contributor who volunteers to upgrade and support a tier 2 board.

Thoughts?

Sympatron, 2021-08-12:
@bradleyharden, I agree with your reasoning. Does that mean everything should stay as it is? I think it would be nice, if a BSP's tier would be obvious, but I don't see how to do that.

Maybe mentioning this in the README, in a new README in `/boards` and/or in the BSPs' READMEs would be enough.

sajattack, 2021-08-12:
Perhaps metro m4 as well?

jbeaurivage, 2021-08-14:
@sajattack, sounds good for `metro_m4`. We would need to get it updated before we pull the trigger on the release

LongLiveCHIEF, 2021-08-14:
I'd be willing to help with the metro_m4 if needed.

sajattack, 2021-08-14:
Thanks chief, please go ahead.

sajattack, 2021-08-22:
@LongLiveCHIEF I had a bit of spare time and took care of the metro m4 BSP.

LongLiveCHIEF, 2021-08-24:
Thanks @sajattack

sajattack, 2021-10-09:
Please update https://github.com/rust-embedded/awesome-embedded-rust with our tier system and new boards :)

jbeaurivage, 2021-10-16:
Yup, mostly done, I think we're just waiting on the pygamer (#455)

TDHolmes, 2021-10-16:
I think closing it ahead of pygamer update is fine. Spun out the update to awesome-embedded-rust to a separate issue as it'd be nice to wait for a few close and in progress BSPs to land first

bradleyharden, 2021-10-16:
I think the wio_terminal was also on the list

TDHolmes, 2021-10-17:
Can we spin out specific issues for those boards? A tier system has been established, so the heart of this issue is definitely complete

bradleyharden, 2021-10-17:
Yeah, that works

closed 2021-10-17 by TDHolmes

bradleyharden, 2021-10-17:
For the record, here are the Tier 1 boards:
- feather_m0
- feather_m4
- metro_m0
- metro_m4
- pygamer
- samd11_bare
- wio_terminal