atsamd-rs/atsamd #436: What type to use to add v2 spi to struct?

cbc02009, 2021-04-26:


bradleyharden, 2021-05-01:
@cbc02009, sorry for the delay. I just saw this.

First, thanks for being a bit of a guinea pig. The v2::spi interface is still very new. I'm trying to work out kinks and better document it.

The first type parameter to the Config struct is the set of SPI pads. That struct needs to implement the Dipodopo trait, which identifies the correct DIPO and DOPO values (are you familiar with those? They are in the SAMD datasheet).

The solution is a bit more complicated than I would like it to be right now. I recently found a way to simplify it, but that change hasn't yet been merged. However, for now, you can use the pads_alias! macro to help you define the type.

https://atsamd-rs.github.io/docs/samd21g/thumbv6m-none-eabi/doc/atsamd_hal/macro.pads_alias.html

cbc02009, 2021-05-02:
@bradleyharden Thanks for the response. I just read through your pull request and I think it handles a lot of the stuff I'm trying to figure out, so maybe I'll wait until that gets mainlined. Once that pull request is finalized and merged in I'd be happy to help updating some of the other examples to follow the same format.

Back to the original issue: my thought was to use the BSPs in a similar way to what the nrf52hal does and put all the pins and peripherals into a single `struct Board` that functions almost like a device tree. This would allow for better auto complete during code writing, since it's a lot easier for IDEs to parse `struct` entries than macro expansions. It would also allow the board to be entirely set up in the main code with a single `let mut board = BSP::Board::take().unwrap()` 

You can see the example I've been working on here (with the nrf52840): https://github.com/cbc02009/adafruit-nrf52840-express-rs. You definitely lose out on configurability with the BSP written this way, as it preconfigures all the peripherals, but I think it makes it more accessible for what most people actually use BSPs for: initial prototyping or simple home projects.

I think once the changes to SPI are made from your pull request it will be quite a bit easier for me to set up this `struct` and see how well it works.

bradleyharden, 2021-05-02:
Interesting perspective on BSPs. Have you seen the discussion in #433? There has been a lot of discussion lately on exactly what role the BSPs should play, how configurable they should be, where they should be stored, and how they should be maintained. Perhaps you could add your thoughts.

cbc02009, 2021-05-02:
Thanks, I'll do that. Again, if you want any help with the easy-to-do grunt work of porting the old BSPs to use your new code let me know. I figure that's at least something I'm capable of helping out with since it's mostly copy-paste + test.

bradleyharden, 2021-06-14:
FYI, I think the final simplification for this issue will land in #451. You can take a look at the feather_m0 in that PR to see how the Spi types are constructed. I also tried to improve the documentation. I now show how to explicitly define the type, in the absence of any type inference.

bradleyharden, 2021-06-23:
@cbc02009, in the latest version, I've added documentation covering how to construct the SPI types. Could you read through it and see if it makes sense? Does it answer your original question?

https://atsamd-rs.github.io/docs/samd21g/thumbv6m-none-eabi/doc/atsamd_hal/sercom/v2/spi/index.html

cbc02009, 2021-06-24:
Yes, thank you. The documentation and the updated feather m0 example have allowed me to instantiate the SPI peripheral within my `Board::new()` function.

I'm not sure if this warrants a new issue, but now that the SPI works, I've run into a second issue with the pins and their ownership. the `bsp_pins` macro takes all the pins and lumps them into one struct, which is not ideal for trying to dole out pins to different peripherals. I'm having difficulty wording this, so let me see if an example would help.

I'm trying to create a `Board` struct that contains fields for all the peripherals on the particular board, like so: 
```
// Provides access to all features of the Adafruit M0 Metro
#[allow(non_snake_case)]
pub struct Board {
    /// The Metro M0's pins which are not being used by peripherals
    pub pins: Pins,

    /// The Metro M0's exposed SPI
    pub ext_spi: spi::Spi<spi::Config<spi::Pads<Sercom4, Miso, Mosi, Sclk>>>,
}
```

`Board` implements `new()` which initializes all the peripherals and returns a `Board` struct, like so:
```
impl Board {
    /// Take the peripherals safely
    ///
    /// This method will return an instance of `Adafruit Metro M0` the first time it is
    /// called. It will return only `None` on subsequent calls.
    pub fn take() -> Option<Self> {
        Some(Self::new(CorePeripherals::take()?, Peripherals::take()?))
    }

    fn new(cp: CorePeripherals, mut p: Peripherals) -> Self {

        // Initialize the clock controller
        let mut clocks = GenericClockController::with_internal_32kosc(
            p.GCLK,
            &mut p.PM,
            &mut p.SYSCTRL,
            &mut p.NVMCTRL,
        );

        //initialize the delay module
        let mut delay = Delay::new(cp.SYST, &mut clocks);

        // take ownership of pins
        let base_pins = Pins::new(p.PORT);

        // Setup the exposed SPI
        let gclk0 = clocks.gclk0();
        let clock = clocks.sercom4_core(&gclk0).unwrap();
        let freq = clock.freq();
        let pads = spi::Pads::<Sercom4>::default()
            .data_in(base_pins.miso)
            .data_out(base_pins.mosi)
            .sclk(base_pins.sclk);
        let ext_spi = spi::Config::new(&mut p.PM, p.SERCOM4, pads, freq)
            .baud(MegaHertz(10))
            .spi_mode(spi::MODE_0)
            .enable();

        Board {
            pins: base_pins,
            ext_spi,
        }
    }
}
```

The problem is that ownership of `base_pins` is now split, so I can no longer return base_pins as a member containing the leftover GPIO pins.

I think what I need to do is define a second struct:
```
pub struct GPIO {
    pub a0: Pin<hal::gpio::v2::PA02, Reset>,
    ...
}
```

and pass that back instead, like:

```
let pins = GPIO {
  a0: base_pins.a0,
  ...
}

Does that sound right? Is there any way I can use the bsp_pins macro to do this for me?

Again, thanks for you help.

bradleyharden, 2021-06-24:
> The problem is that ownership of base_pins is now split, so I can no longer return base_pins as a member containing the leftover GPIO pins.

Correct

> I think what I need to do is define a second struct

Also correct.

Indeed, it's a pin of a pain. I'm not sure if there is a macro solution for it. I'll try to think about it.

bradleyharden, 2021-06-26:
@cbc02009, since your original question has been answered, I'm going to close this. If you want to discuss the pin issue further, please open a new issue or join us on Matrix.

closed 2021-06-26 by bradleyharden