atsamd-rs/atsamd #439: Question about atsamd51 ADC.

ewwaller, 2021-05-08:
I am new to rust, so this may be part of the learning curve, but I cannot figure out how to get an instance of an ADC on the samd51j?   I can get the example for the samd21 to build, but the demo is missing for the 51j (Feather_m4 board)

It looks like I need to use the adc_hal macro, but it is not exported.  Attempts to use ADC0 or ADC1 report they are private.  I cannot see any way to get an initialized structure.

Can someone provide a pointer?   Or am I missing something really basic?

sajattack, 2021-05-08:
There is a 51j adc example for the metro m4 https://github.com/atsamd-rs/atsamd/blob/master/boards/metro_m4/examples/adc.rs

ewwaller, 2021-05-08:
Thanks.  I'll work through that and get back.

ewwaller, 2021-05-11:
Thank you sajattack.  I was close, but that helped a great deal.  The underlying problem was I had created a mess with dependencies I created by being unsure what to use directly, and what to use through through other modules (I had been trying to use atmsd51j directly for another reason).  In any event, having a working reference allowed me to untangle the rest of my mess.

The feather M4 board includes a voltage divider from the battery voltage to the 'a6' pin which is not broken out on the board.  I forked this project to add that pin to the M4 support package.  As long as I am in there, perhaps I will put together an ADC demo before I submit a merge request.

closed 2021-05-11 by ewwaller

pustekuchen91, 2023-11-30:
Hey, i have a question about the adc. I'm trying to read an adc pin on a ATSAME53N20A.

# My Test code

```
#[embassy_executor::task]
pub async fn mytest() {
    let mut peripherals = Peripherals::take().unwrap();
    let pins = myboard::bsp::pin::Pins::new(peripherals.PORT);
    let _core = CorePeripherals::take().unwrap();
    let mut clocks = GenericClockController::with_external_32kosc(
        peripherals.GCLK,
        &mut peripherals.MCLK,
        &mut peripherals.OSC32KCTRL,
        &mut peripherals.OSCCTRL,
        &mut peripherals.NVMCTRL,
    );

    let mut adc1 = Adc::adc1(
        peripherals.ADC1,
        &mut peripherals.MCLK,
        &mut clocks,
        GEN_A::GCLK1,
    );

    let mut hardware_info_pin = pins.hardware_info.into_alternate::<B>();

    let data: u16 = adc1.read(&mut hardware_info_pin).unwrap();
    debug_rprintln!("Value: {}", data);
}
```

# Error
```
13:44:59.863 panicked at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/atsamd-hal-0.16.0/src/thumbv7em/adc.rs:268:1:
13:44:59.863 adc clock setup failed
```