atsamd-rs/atsamd #440: EIC reference design or example?

ewwaller, 2021-05-15:
I am attempting to use an external interrupt on PA14 on an atsamd51j.  This is an rtic project.

The eic is defined in the resources struct as :
```
gpioInput: eic::pin::ExtInt14<feather_m4::gpio::Pin<gpio::v2::PA14, hal::gpio::v2::Alternate<feather_m4::gpio::v2::A>>>,
```
and is initialized as:

```		
let gpioInput = pins.d4.into_pull_down_input(&mut pins.port).into_ei(&mut pins.port);
```

The hardware task is
```	#[task(binds = EIC_EXTINT_14, priority = 5, resources = [gpioCount])]
    fn gpio_interrupt(cx: gpio_interrupt::Context) {
		*cx.resources.gpioCount += 1;
	}
```
It compiles and runs, I can see valid pulses with a scope on d4 (PA14).  But, no interrupts are captured. 

I am sure I am missing a step.  Is there a reference design or example code for setting up the samd51 for external interrupts?

ianrrees, 2021-05-18:
Hi @ewwaller , sorry I don't have a SAMD51 board to test with, nor experience with RTIC, but there's an EIC example [here](https://github.com/atsamd-rs/atsamd/blob/master/boards/trinket_m0/examples/eic.rs) for SAMD21 which may help.  Additionally, I think you'll need to unmask the EIC interrupt:

``` Rust
unsafe {
    NVIC::unmask(interrupt::EIC);
}
```

And possibly add an ISR:
``` Rust
#[interrupt]
fn EIC() {
    defmt::info!("In EIC ISR");
}
```

That said, I came here after running in to trouble enabling a pullup and interrupt on the same GPIO pin.  It seems the issue is that `into_ei()` call, which in my case is calling `.into_function()` and disabling the pullup.

ewwaller, 2021-05-19:
Thanks Ian,
That is pretty close to what I had.  I have it working now, but have a few unanswered question that I still need to resolve. 

The atsamd51j hal does not implement eic::init  (at least not as a public function);  it does implement eic::init_with_ulp32k which yields a ConfigurableEIC rather than an EIC.  That ConfigurableEIC provides the expected methods of sense, enable_event, enable_interrupt, etc.   But, the Configurable EIC does nothing until it is ""Finalized" by a call to the finalize method, after which interrupts are processed. All good.   I assert the documentation on all of this is a bit thin.   I imagine the init_with_ulp32k configured the eic so it works when the processor is sleeping. 

Problems remain, however.  For instance, how does one disable the interrupts?  The disable_interrupt method seems to only work on the ConfigurableEIC, but will not work with it after it has been finalized and converted to EIC.

Another problem is how this fits into the RTIC framework.   If one sets up the EIC in the init section, the pin (which has been converted into an eic) needs to be passed into the resources structure.   This is because the EIC_EXTINT_xx ISR needs reference to the eic to clear the interrupt.  Since it is necessary to enable the interrupt before it is finalized, and the eic is finalized before it gets passed into the resources structure as a LateRosources, is there not an instant between where the interrupt is enabled and the reference to the pin in the resource structure has not been initialized?  Would not an interrupt during this period be catastrophic?     I suppose it is possible that the init section of RTIC is protected as a critical section with interrupts disabled -- can someone verify whether this is the case?

bradleyharden, 2021-06-14:
@ewwaller, where did you land with this? It looks like #444 will land soon. I think you needed pull resistors as well. But it also sounds like there is other functionality missing. Feel free to make a pull request.

bradleyharden, 2021-06-26:
@ewwaller, now that #444 has landed, I'm going to close this. Feel free to reopen if needed.

closed 2021-06-26 by bradleyharden