atsamd-rs/atsamd #442: Arduino Nano 33 IoT: Serial example not working

DCRichards, 2021-05-19:
I am currently unable to get the [serial.rs](https://github.com/atsamd-rs/atsamd/blob/master/boards/arduino_nano33iot/examples/serial.rs) example to work as intended. I am able to compile and deploy to the board but the board is then not recognised as a USB device, so I am unable to communicate with it unless I put it into bootloader mode, at which point is visible as a USB device again.

I am using this with a [Vagrant setup](https://github.com/DCRichards/arduino-vagrant-rust) but the board is also not recognised locally on my mac (Mac OS 10.15.7) so I believe this is an issue with the example code.

DCRichards, 2021-05-19:
@sajattack @nebelgrau77 I believe you built this example so may be able to offer some insight?

Thank you to all those who have been working on this by the way, really great work.

sajattack, 2021-05-19:
This is uart serial not usb.

sajattack, 2021-05-19:
Try adapting this example: https://www.github.com/atsamd-rs/atsamd/tree/master/boards%2Fmetro_m0%2Fexamples%2Fusb_serial.rs

closed 2021-05-19 by sajattack