atsamd-rs/atsamd #446: USB freezing

Cightline, 2021-05-26:
Symptoms: `lsusb` on Linux will freeze temporarily when I restart or plug in my Feather M0. The freeze lasts about 10-20 seconds. 10% of the time when it unfreezes the Feather M0 does not appear in the `lsusb` output. I am using RTIC so I suspect it may be related. 

@ianrrees mentioned in chat that I could sniff the traffic with Wireshark. It may be a couple days before I can do that, so please standby.

TDHolmes, 2021-09-26:
@Cightline any update on this one? Wondering if any changes since then might have fixed this

TDHolmes, 2021-10-17:
This seems unlikely to be specific to our HAL code, and since we haven't heard an update, closing

closed 2021-10-17 by TDHolmes

ianrrees, 2022-02-21:
This looks similar - I intend to dig in to soon: https://github.com/mvirkkunen/usb-device/issues/84

ianrrees, 2022-03-12:
Aha!  @Cightline if you've got a minute, could you test with this modification to the HAL?  https://github.com/atsamd-rs/atsamd/pull/579

Cightline, 2022-03-24:
I'll have time this weekend, been busy.