atsamd-rs/atsamd #447: When is the master usuallly being released to crates.io?

MangoIV, 2021-05-31:
As the title states, I would like to know how long it will last for the next version to come out because I would like to use some changes made on the master with my current project. 
Thanks in advance!

bradleyharden, 2021-05-31:
Yeah, I think it's been quite a long time since the last release. I think that's been motivated by a lot of the significant changes going on. There's been an effort to restructure some of the modules (#426), rethink BSPs (#433), breaking changes to some of the new v2 APIS, and updates of some BSPs to use the new v2 APIs.

We should probably put together a release soon. Do you know about our Matrix channel chat? We'll probably discuss the issue there as well.

MangoIV, 2021-05-31:
Where would I finde the matrix chat? Would be interested in that... And yeah, I noticed that the v2 stuff isn't used in the examples which gave me some trouble, honestly :D 
Thanks for your quick answer!

bradleyharden, 2021-05-31:
Check out the feather_m0 on master. It has been fully ported to v2. We just merged that yesterday.

Matrix is a chat protocol. I think the most common client is [Element](http://element.io/). Once you get set up, find is in the atsamd-rs channel.

sajattack, 2021-05-31:
matrix.to/#/#atsamd-rs:matrix.org

closed 2021-06-03 by MangoIV