atsamd-rs/atsamd #463: itsybitsy_m4/examples/usb_serial.rs is broken - older version works

billyrayvalentine, 2021-06-29:
usb_serial.rs in master seems to be broken (at least on the ItsyBitsy M4 Express) - that is: no debug output comes out of the UART and no ACM device is enumerated with lsusb

However, if i checkout https://github.com/atsamd-rs/atsamd/blob/15d084c99b5cfed81b71d2d6f29123f47c2baf6f/boards/itsybitsy_m4/examples/usb_serial.rs this version works fine: device is shown in lsusb, debug messages are dumped over the UART and can set the led colour over the ACM device etc.

I can see in the roadmap the UARTv2 may replace this, so this may be a  won't fix item, but raising anyway.

Using:
```
rustc --version 
rustc 1.53.0 (53cb7b09b 2021-06-17)
```
and ItsyBitsy M4 Express

Hope this helps

ianrrees, 2021-06-30:
Hi @billyrayvalentine - I don't have a SAMD51 board to test with at the moment, but skimming over that example I don't see any reason why it shouldn't work.  On SAMD21, I've noticed that the USB gets a bit flaky if the firmware is built in debug mode, so perhaps try `cargo build --release --example=usb_echo`.

The UARTv2 is about the UART hardware, and the usb_serial example seems to build a hardware UART.  However, I think it actually just echoes over the USB virtual serial port - looks a bit like it is a work-in-progress.

ianrrees, 2021-07-06:
I've now got a Metro M4 to play with; a super quick test attempting to jam that itsybitsy M4 example in didn't work, but I'm not sure it should've.  Will take a closer look in the next few days, but in the mean time I can confirm that the Metro M4 usb_logging example does work for me.

TDHolmes, 2021-09-26:
Given the mentioned issue/PR is closed and we couldn't reproduce ourselves anyways, should this be closed?

TDHolmes, 2021-11-29:
@Josfemova / @billyrayvalentine , I've been seeing this issue as well and it seems I get correct behavior again after disabling the data and instruction cache:
```rust

    // disable all caching
    unsafe {
        peripherals.CMCC.ctrl.write(|r| r.bits(0_u32));
    }
```

I'm not sure if this is a generic code bug in our HAL or the `usb-device/usbd-serial` crates, or if we need to be issuing specific cache clear calls ourselves. Either way, this gives me more consistent behavior for now...

Josfemova, 2021-11-29:
is there a "magic place" for that line?
I tried it and still can't get the correct behavior

TDHolmes, 2021-11-29:
I placed it at the very top.
```rust
#[bsp::entry]
fn main() -> ! {
    let mut peripherals = Peripherals::take().unwrap();
    let mut core = CorePeripherals::take().unwrap();

    // disable all caching
    unsafe {
        peripherals.CMCC.ctrl.write(|r| r.bits(0_u32));
    }
```

However, it looks like it should already be disabled by default on boot and I don't see us enabling it in the HAL. In my debugging of another issue, I also tried disabling write buffering with:
```rust
    // disable write buffer (DISDEFWBUF) to get a precise hard fault
    unsafe {
        core.ICB.actlr.modify(|v| v | 2_u32);
    }
```
I wonder if that was still in effect.

TDHolmes, 2021-11-29:
for me, if I do either of these commands at the top, my USB enumeration woes go away. Without it, I can see it enumerated incorrectly with corrupted strings.

Josfemova, 2021-11-29:
Ok so the ICB stuff made me "discover" the example is pulling version 10 of the pac, so I will fight with dependency hell for some time  and come back I guess

TDHolmes, 2021-11-29:
I take it back, data and instruction caching is enabled by default for SAMD5x/SAME5x, so there does seem to be a data caching issue somewhere in here. Wouldn't see it on SAMD2x/1x because they don't have caching IIRC. ` peripherals.CMCC.ctrl.write(|r| r.bits(0_u32));` should disable all caching.

re: v0.10.0 of the PAC, that shouldn't really impact this specific usage I would think.

Josfemova, 2021-11-29:
About de PAC: ICB is not in version 0.10, at least not in CorePeripherals. It is there on version 0.11 tho

TDHolmes, 2021-11-29:
I changed something else and I'm back to broken on either of these changes. I think this is all a red herring :/

Josfemova, 2021-11-29:
So, using PAC v0.10 disabling CMCC does nothing...but on v0.11 the builtin led starts on color purple.
Still not getting this thing to  work, but at least now dmesg says something
```
[34097.439917] usb 1-5: new full-speed USB device number 55 using xhci_hcd
[34097.580892] usb 1-5: unable to get BOS descriptor or descriptor too short
[34097.581935] usb 1-5: unable to read config index 0 descriptor/start: -32
[34097.581942] usb 1-5: chopping to 0 config(s)
[34097.581946] usb 1-5: can't read configurations, error -32
[34097.703218] usb 1-5: new full-speed USB device number 56 using xhci_hcd
[34097.843893] usb 1-5: unable to get BOS descriptor or descriptor too short
[34097.844866] usb 1-5: unable to read config index 0 descriptor/start: -32
[34097.844869] usb 1-5: chopping to 0 config(s)
[34097.844870] usb 1-5: can't read configurations, error -32
[34097.844898] usb usb1-port5: attempt power cycle
[34098.246529] usb 1-5: new full-speed USB device number 57 using xhci_hcd
[34098.264400] usb 1-5: unable to get BOS descriptor or descriptor too short
[34098.265458] usb 1-5: unable to read config index 0 descriptor/start: -32
[34098.265465] usb 1-5: chopping to 0 config(s)
[34098.265469] usb 1-5: can't read configurations, error -32
[34098.386516] usb 1-5: new full-speed USB device number 58 using xhci_hcd
[34098.404465] usb 1-5: unable to get BOS descriptor or descriptor too short
[34098.405533] usb 1-5: unable to read config index 0 descriptor/start: -32
[34098.405541] usb 1-5: chopping to 0 config(s)
[34098.405545] usb 1-5: can't read configurations, error -32
[34098.405616] usb usb1-port5: unable to enumerate USB device
```

Josfemova, 2021-12-07:
Ok so update:
Without changing the `usb_serial` program, things now...work (?
How? Welp, updating cargo-hf2/hf2-cli makes it work.
Seems to be related to [this](https://github.com/jacobrosenthal/hf2-rs/pull/44) but idk. 
@billyrayvalentine could you check if this solves the issue for you?

TDHolmes, 2021-12-08:
This fixed the issue for me too!

bradleyharden, 2021-12-12:
Can we close this then?

TDHolmes, 2021-12-12:
I think so

closed 2021-12-12 by TDHolmes