atsamd-rs/atsamd #464: gpio::v1: Faulty inital state of pins returned from `PORT::split()`

vccggorski, 2021-07-01:
By design, all GPIO pins are in a `FloatingDisabled` (aka `Reset`) state. Because of backward compatibility in `gpio::v1`, initial state is incorrectly assumed as `FloatingInput`. `v2::Pins::into_mode` based routines make an additional check if mode is in fact changed, otherwise it is a noop. `v1::Pins` are using `v2::Pins` internally. Because of misattributed initial mode, it is impossible to change a mode of a pin to `FloatingInput` as it is incorrectly assumed to be `FloatingInput` already.

Proposed solution: Create a similar `into_mode` mechanism inside of a `gpio::v1` module that ignores a check. It will make a behaviour a compatibility shim at least a little bit more sane.

Tagging @bradleyharden

bradleyharden, 2021-07-02:
@vccggorski, try this patch. I can merge it myself later, once you've had a chance to test it.

https://gist.github.com/bradleyharden/52696ccddb8c783f5f4885fb45162b3e

vccggorski, 2021-07-02:
As you already know, I moved to v2 implementation of both gpio and sercom. For the sake of testing this patch, In order to test this patch I hacked it back together and it seems like it has fixed an issue.

closed 2021-07-29 by jbeaurivage