atsamd-rs/atsamd #465: Framework for an eic::v2 module

bradleyharden, 2021-07-02:
@ianrrees brought up some limitations of the current `eic` module on Matrix. I suggested that he might consider writing an `eic::v2` module. Here is the type-level framework that I was imagining for it. Anyone should feel free to pick this up and run with it. I can advise, but I don't have the bandwidth to flesh it out anymore than this.

https://gist.github.com/bradleyharden/09a72714dde9f4dd1122834c6538abf8

bradleyharden, 2021-07-02:
Hmm. I just realized a flaw with this. You can only have one clock source for the entire EIC. You can't have an individual one for each `ExtInt`. So the `EIController` needs to be generic over an `OptionalEIClock` or something. But each `ExtInt` still needs to be generic over `DetectionMode`.

vcchtjader, 2021-10-15:
I had a go at this, see https://github.com/vccggorski/atsamd/pull/2

It is based on Clocking v2 so that presents some complications in regards to creating a PR for easy reviewing, not sure what is the best way forward here, but as it retains v1 and v2 is opt-in.

I use the "Enabled<...>" from clocking to keep track of ExtInt[0..16], but using it directly from clock::v2 makes the documentation end up together with clocking which is far from optimal, so that is for now a copy.

I focused on `thumbv7`, but found that thumbv6 is a subset of the v7 features which I think would be possible to express via feature flags.