atsamd-rs/atsamd #475: Improving the `bsp_pins!` macro

bradleyharden, 2021-07-25:
@ianrrees, I mentioned creating an issue in one of your PRs. Here it is. I'd also like to ping @anall, and @quentinmit specifically, since they have shown an interest in the topic previously.

There has been some discussion about the `bsp_pins!` macro and some possible improvements. I wanted to lay out the problem and show the potential changes I've considered. Let me know what you think.

# Introduction

First, I want to start with a little primer, for those not intimately familiar with the `gpio::v2` module.

There are two APIs: `pin` and `dyn_pin`. The `dyn_pin` module uses an approach that should be straightforward for anyone familiar with Rust. It defines the `DynPin` type in terms of `DynPinId` and  `DynPinMode` types, which are themselves defined in terms of `DynGroup` and other `Dyn*` types, like so:

```rust
pub struct DynPinId {
    pub group: DynGroup,
    pub num: u8,
}

pub enum DynGroup {
    A,
    B,
    C,
    D,
}

pub enum DynPinMode {
    Disabled(DynDisabled),
    Input(DynInput),
    Interrupt(DynInterrupt),
    Output(DynOutput),
    Alternate(DynAlternate),
}

pub enum DynInput {
    Floating,
    PullDown,
    PullUp,
}

// ...

pub struct DynPin {
    id: DynPinId,
    mode: DynPinMode,
}
```

The `pin` module elevates this from a value-level API to a type-level API. Each `Dyn` `enum` becomes a trait, and each variant becomes a type. Then, the `Pin` type is defined in terms of the `PinId` and `PinMode` traits.

```rust
pub trait PinId {
    /// Corresponding [`DynPinId`]
    const DYN: DynPinId;
}

// The PinId trait is not split into PinGroup and PinNum traits, although it could be

pub trait PinMode {
    /// Corresponding [`DynPinMode`]
    const DYN: DynPinMode;
}

pub trait InputConfig {
    /// Corresponding [`DynInput`]
    const DYN: DynInput;
}

pub enum Floating {}
pub enum PullDown {}
pub enum PullUp {}

// ...

pub struct Pin<I, M>
where
    I: PinId,
    M: PinMode,
{
    id: PhantomData<I>,
    mode: PhantomData<M>,
}
```

# Problem

The `bsp_pins!` macro loops over each `PinId` type, e.g. `PA24`, and defines various identifiers, type aliases and constants. For instance, a block for `PA24` might look like this.

```
PA24 {
    name: miso,
    aliases: {
        AlternateC: Miso,
        AlternateD, UartRx,
        AlternateH: UsbDp,
    }
}
```

For the most part, this has seemed to work well. However, there are a few shortcomings I would like to discuss.

## `Pins` struct

Right now, we create a `Pins` struct that contains the complete collection of all pins for a chip. Each field of this struct is `pub`, so users can partially move the pins they need out of the struct and then drop the rest. This approach works well, but it has one major downside; you can only give each pin a *single* field name. Because of that, you end up referring to `pins.miso`, even if you want to use the pin as `UartRx` or `UsbDp`.

There are alternatives to this approach, but each has downsides:

- We could wrap each pin in an `Option` inside the `Pins` struct. Then we could provide multiple methods that return the same `Option<Pin>`. However, if you try to use the same pin for two different purposes, a previously compile-time error now becomes a run-time panic, which is not ideal. Moreover, now that the zero-sized pin structs are wrapped in `Option`, they become sized, so the `Pins` struct ends up being quite large (probably at least 20+ bytes, depending on how many pins your chip has).
- We could use type-level programming and have the `Pins` struct hold a collection of `OptionalPin`s, i.e. a pin or `NoneT`. This is the ideal solution from a technical perspective. You still get compile-time errors, and the `Pins` struct is still zero-sized. But it has pretty terrible ergonomics. First, it would require the `Pins` struct to have 20+ type parameters. And second, every time you take a pin out of the `Pins` struct, you have to change one of the `Pins` type parameters from a pin to `NoneT`. That means each `Pins` method would have to take `self` and return a modified `Pins` struct, along with the desired pin.
- We could come up with some creative macro solution to translate all three of `pins.miso`, `pins.uart_rx` and `pins.usb_dp` to `pins.pa27`. This is probably the least intrusive way to do it, but it adds a layer of macro indirection to an otherwise simple task.

Alternatively, we could do nothing and keep the status quo. It just means people need to know and understand that each pin has multiple uses and names, and that one of the names is the "default".

## `Dyn` constants

Right now, there are two different `Dyn` constants defined for each alias above. The macro produces something like this.

```rust
pub const UART_RX_ID: DynPinId = <PA24 as PinId>::DYN;
pub const UART_RX_MODE: DynPinMode = <AlternateD as PinMode>::DYN;
```

However, because the type-erased, value-level, `dyn_pin` API is not used very often, I can't say whether these constants are sufficient.

## Type aliases

The type-level API is more commonly used, so I have been able to identify some improvements.

Right now, only one type alias is defined for each alias above:

```rust
pub type UartRx = Pin<PA24, AlternateD>;
```

This is probably the most useful type alias, but there could be others. For instance, we could define something like

```rust
pub type UartRxId = PA24;
pub type UartRxMode = AlternateD;
```

Alternatively, we could define a generic alias for the pin in any mode:

```rust
pub type UartRxPin<M> = Pin<PA24, M>;
```

That would allow you to refer to a pin by multiple names, no matter the `PinMode`.

Are any of these useful? Are there any other ideas to add? What about the `name` field of the macro (in contrast to the `aliases` block)? Do we want to generate anything based on it?

quentinmit, 2021-07-26:
I haven't used the dynamic pin API at all, because compile-time checking seems like a much better idea to me.

I wonder how much mileage we could get just by having the convention that pin names contain all the aliases? e.g. instead of calling the pin `miso`, what if you called it `pa24_miso_uartrx_usbdp`? It's verbose, but it's also clear. That also avoids the possibility of collision (if someone tried to use `miso` and `uartrx` at the same time, without realizing they're the same pin).

The `Id` and `Mode` type aliases seem redundant to me, since you can already do `<UartRx as AnyPin>::Id`. I wonder if it's possible to expose `::Id` directly on each `Pin` as well, to avoid needing to cast it to `AnyPin`.

BTW, sorry I haven't been participating much lately. Life has gotten in the way.

jbeaurivage, 2021-07-26:
Another option could be to devise some kind of macro system that translates pin name aliases. For example, let's assume the default pin name: is `uart_rx`:
```
let mosi = pin_alias!(pins.mosi);
```

The macro would do the conversion:
```
pin_alias!(pins.mosi) -> pins.uart_rx
```

The default pin name wouldn't need to be translated through the macro, although maybe we could require it just for symmetry's sake. 

I don't know if that's practical or a good idea though.

ianrrees, 2021-07-27:
The type aliases like `pub type UartRxId = PA24;`, were what I was thinking about in that initial conversation.  I imagine using them when declaring statics, especially in projects where there might be a couple BSPs in play.  That situation has come up in my work project - I've been developing against a board that is wired a bit differently from what the real product will be, and it would be nice to just switch the BSP from the development one to the real hardware one without having to change any types in the main source.

Is there any downside in the macro generating more type aliases?

All that said, the only situation where that occurs is where I'm storing a pin ID in an enum to implement open drain output, as discussed in https://github.com/atsamd-rs/atsamd/issues/466 .  If we had an open drain mode in the HAL, I could use the type alias that bsp_pins! currently generates.  And, it's totally not a big issue for me; just a one line change, and I'll likely not be going back to development hardware once the real thing is in hand.

bradleyharden, 2021-07-27:
# Responses

@quentinmit

> BTW, sorry I haven't been participating much lately. Life has gotten in the way.

No worries

> what if you called it `pa24_miso_uartrx_usbdp`?

I don't really like this approach. It's verbose, which isn't great, but it's also brittle. Adding a new alias, even if you don't use it, would break all of your code. If users want this approach, I would rather let them do it manually.

> The `Id` and `Mode` type aliases seem redundant to me, since you can already do `<UartRx as AnyPin>::Id`.

That's true, but I don't think redundancy is necessarily bad. The alias would definitely be more succinct and readable. Plus, it doesn't require understanding the `AnyPin` trait.

> I wonder if it's possible to expose `::Id` directly on each `Pin` as well, to avoid needing to cast it to `AnyPin`.

I mean, `Id` effectively **is** exposed directly on each `Pin`.

It sounds like you want types themselves to have associated types, rather than traits. But that's not possible. The only way to map from one type to another is through a trait. And you have to tell the compile *which* trait somehow. In a context `where P: AnyPin`, the compiler can infer it. But absent that context, there's no way to get the corresponding `Id`.

@jbeaurivage

> Another option could be to devise some kind of macro system that translates pin name aliases.

Yup, that's exactly what I was thinking. I like the name `pin_alias!`. I wasn't sure what to call such a macro.

@ianrrees

> Is there any downside in the macro generating more type aliases?

I don't think so. It can generate a lot of clutter in the documentation. But that's easy enough to solve by putting the `bsp_pins!` call within its own module. I did that for the pygamer port. I'm not sure if I did that for other BSPs.

> I'm storing a pin ID in an enum

I'm confused by this. How are you storing a `PinId`? I believe all of the `PinId`s are defined as `enum PXXX {}`, which means they can't be instantiated; they only exist at the type level. Are you using `PhantomData` to "store" the `PinId`?

# Proposal

Here's an idea. What if we simply gave users more options?

- First, I could make the `name` field optional. If not provided, the field name would default to the `PinId` name, i.e. `pins.pa24`.
- Separately from that, I could also add a `pin_alias!` macro definition that would act as @jbeaurivage describes. It would translate any alias name to the correct field name, whatever that may be.
- Finally, I could also add the `Id` and `Mode` type aliases, just to make life easier.

My only concern with this is documentation. It might be difficult to explain all of the different `bsp_pins!` features. But if we use things like the `pin_alias!` macro in our examples, maybe that will help.

Thoughts?

ianrrees, 2021-07-27:
> > I'm storing a pin ID in an enum
> 
> I'm confused by this. How are you storing a `PinId`? I believe all of the `PinId`s are defined as `enum PXXX {}`, which means they can't be instantiated; they only exist at the type level. Are you using `PhantomData` to "store" the `PinId`?

Sorry, imprecise language:
``` Rust
enum NegativeOpenDrainReset<I: PinId> {
    Asserted(Pin<I, PushPullOutput>),
    Deasserted(Pin<I, PullUpDisabled>),
}
```

Thanks @quentinmit for mentioning `<UartRx as AnyPin>::Id` - somehow I hadn't appreciated that until just now!

> It might be difficult to explain all of the different bsp_pins! features. But if we use things like the pin_alias! macro in our examples, maybe that will help.

Yeah, this seems like the key.

bradleyharden, 2021-08-09:
Small update. I was able to implement all three of the proposed points above. It took a little bit of macro magic, but I think it turned out well. The only thing left to do is update the documentation. I think that will be difficult, though, given the list of features.

closed 2021-10-17 by TDHolmes