atsamd-rs/atsamd #488: sleeping_timer_rtc examples don't work

TDHolmes, 2021-09-20:
According to git bisect, the culprit is a783ba9248f3130baf38c910b8ab1ebb6bbd66e2. Will work on fixing this

TDHolmes, 2021-09-20:
ping @henrikssn since it's your commit, but I'll be taking a look at how to fix soon so don't feel obligated to investigate if you don't have the time

closed 2021-10-02 by twitchyliquid64

ryankurte, 2021-10-03:
hey @TDHolmes what did you test this with? i've been trying to get the RTC running with RTIC on an m4 and am have having no luck today

TDHolmes, 2021-10-03:
Hmm, shoot I thought I had tested the RTIC tests too. I'm mostly using `feather_m0` and looking at `sleeping_timer_rtc`

TDHolmes, 2021-10-04:
@ryankurte I played around with this a bit today and couldn't figure it out. Can you create a new issue tracking this regression? I'm confused how me changing the implementation of the embedded-hal timer traits affected rtic-monotonic, because as far as I can tell they don't seem linked.

PR #490 (2021-10-02): Fix RTC implementation of embedded-hal timer traits to be periodic again
In the RTC refactor #412, the RTC timer implementation behavior changed to no longer be periodic, in spite of the embedded-hal Periodic marker trait still being implemented. this restores the periodic behavior of the implementation. Closes #488