atsamd-rs/atsamd #501: RTC broken on SAMD51G

ryankurte, 2021-10-04:
hey there,

i've been messing around with trying to get the RTC (and RTIC) working on a `SAMD51G19A` with (almost) no success. attempting to use `atsamd-hal@master` and `cortex-m-rtic@v0.6.0-alpha.5`. possibly related to: #488.

i started with the setup from [feather_m4/examples/sleeping_timer_rtc.rs](https://github.com/atsamd-rs/atsamd/blob/master/boards/feather_m4/examples/sleeping_timer_rtc.rs#L40) which doesn't seem to actually kick off the RTC. looking at the datasheet the `count32_mode` function seems to be missing some steps, and there's a significant delta between [`Rtc::count32_mode`](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/rtc.rs#L224) and [`Rtc::into_count32_mode`](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/rtc.rs#L160), in fact if one adds this line the RTC does start to function:

```rust
let rtc = Rtc::count32_mode(peripherals.RTC, 1024.hz(), &mut peripherals.MCLK);
let mut rtc = rtc.into_count32_mode();    // Adding this line starts the RTC ticking
rtc.enable_interrupt();
```

so it's possible that was just missed in a previous refactor.

from there i've been attempting to use the RTC as a [`Monotonic`](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/rtc.rs#L468) source for RTIC per [feather_m0/examples/blinky_rtic.rs](https://github.com/atsamd-rs/atsamd/blob/master/boards/feather_m0/examples/blinky_rtic.rs#L38). one can observe that the RTC is running via increasing values from `monotonics::now()`, scheduling immediately (via `::spawn()`) works so the _dispatcher_ would seem to be okay, but `::spawn_after()` does not. attaching a debugger i can see the first call, but can't seem to step into the scheduling to see what it _thinks_ it's doing.

[image]

it's likely i've missed or misunderstood something but, would definitely appreciate any advice y'all have ^_^

ryankurte, 2021-10-05:
swapped to `SysTick` based monotonic and things get scheduled, so, it appears it's something to do with the RTC and/or embedded-time.

sajattack, 2021-10-05:
@TDHolmes I feel like you're the most likely person to have an idea what the issue could be.

TDHolmes, 2021-10-06:
Yeah pretty confident my most recent RTC change broke our monotonic RTIC
implementation. I tried looking at this last weekend and couldn’t figure it
out before I ran out of time. I can look st this again Saturday

TDHolmes, 2021-10-06:
Ping @henrikssn too

AfoHT, 2021-10-11:
For both SAMD51 and SAMD21 I think one of the features of the `OSCULP32K` is that it is always on. At least on the SAMD51 it is only possible to mask the outputs, even though the oscillator is always on

From the datasheet for SAMD21, page 161: 

[image]

jonapap, 2022-12-07:
I've recently encountered the exact same issue on the SAME51J20A. The RTC monotonic clock doesn't work, and similarly adding breakpoints allows it to run.

For anyone else that encounters this, one workaround is to add a delay in either `set_compare` or `clear_compare_flag`. For example:
```rust
    fn clear_compare_flag(&mut self) {
        self.mode0().intflag.write(|w| w.cmp0().set_bit());

        for i in 1..10000 {
            nop();
        }
    }
```

Obviously, this is far from ideal. However, with this above change, everything works fine, including going to deep sleep along with `wfi`.

My (uneducated) guess is that some weird synchronization issue is happening here. However, the datasheet makes no mention of needing synchronization for either `intflag` or `comp0`. I've found some places ([here](https://community.arm.com/support-forums/f/keil-forum/27856/cortex-m3-nvic-when-to-clear-the-interrupt-flag) and [here](https://developer.arm.com/documentation/ka003795/latest)) that talk about similar issues. I've tried various things, such as calling `dsb()` and `lsb()`, without success. If anyone else has an idea what could be causing this, I would be more than happy to test.