atsamd-rs/atsamd #503: Using [package.metadata] in each crate instead of maintaining a Crates.json file

LongLiveCHIEF, 2021-10-04:
what do you guys think of using the `[package.metadata.tier]` format in the `Cargo.toml` so we can get rid of having to maintain a `Crates.json` file?

It would make things much simpler, especially as we move into auto-generating `CHANGELOG` and even auto-updating `README`with list of tiered crates

likewise for `[package.metadata.build_invocation]`
each crate would add those two metadata fields, and specify their tier and build invocation, and then I can use that in the build matrix to modify the README accordingly as well as invoke any build instructions
https://doc.rust-lang.org/cargo/reference/manifest.html#the-metadata-table

This would also easily allow the build stage to check to ensure any new crates have the appropriate information required in the Cargo file, enabling us to dynamically add crates (especially BSP's) very easily.

LongLiveCHIEF, 2021-10-04:
maybe just a single `[package.metadata.atsamd_metadata]`:

```
[package.metadata.atsamd_metadata]
crate_type = BSP # BSP | PAC | HAL
tier = 2 # only applicable for 
build_invocation = "cargo build --examples --features=unproven,usb"
```

jbeaurivage, 2021-10-04:
This makes perfect sense to me!

LongLiveCHIEF, 2021-10-04:
related #481

LongLiveCHIEF, 2021-10-04:
Proposal spec:

---

### Adding a new BSP/PAC

The build and release workflows will automatically recognize any new crate added to the repo,
but you will need to add some metadata to the crates's `Cargo.toml`.

```toml
# example of metadata for a BSP
[package.metadata.atsamd]
# also_build (optional): indicate if any additional crates (HAL | PAC | BSP's) should be built
# to properly test this crate
# also_build = ["crate", "crate2"]
# room for other generic high-level hooks for pipeline/build customization

[package.metadata.atsamd.bsp]

# tier (required)
# see: https://github.com/atsamd-rs/atsamd#how-to-use-a-bsp-ie-getting-started-writing-your-own-code
# for determining what tier your BSP is
tier = 2

# build_command (required)
# used to invoke the appropriate build command for the bsp
build_command = "cargo build --examples --features=unproven"

[package.metadata.atsamd.pac]
# target (required): the build target, used to configure the runner properly
target = "thumbv6m-none-eabi"
# features to build the pac with
features = ["samd21g", "unproven", "usb"]
```

sajattack, 2021-10-04:
Please update the wiki pages related to adding new boards/pacs.

LongLiveCHIEF, 2021-10-05:
This is also in a new `CONTRIBUTING.md` in the branch I'm working on: https://github.com/LongLiveCHIEF/atsamd/blob/use-package-metadata/CONTRIBUTING.md, which makes it pop up during merge requests and other github integrations.  (I've also updated the merge request template)