atsamd-rs/atsamd #517: Automated Comment in PR when HAL or PAC causes T1 build failure

LongLiveCHIEF, 2021-10-17:
This is part of the effort for #481 

Per recent discussions, T1 boards will be built when PR's to HAL and associated PAC's are submitted.  It will be the responsibility of the contributor to ensure all T1 board builds pass.

On the event of a T1 board build failure, there should be a workflow that leaves a comment that notifies the PR author that their change caused a T1 build failure, and that it is their responsibility to make those builds pass before the PR can be merged.

- [ ] update contributing documentation to reflect responsibilities of the contributor in regards to T1 boards on HAL/PAC changes
- [ ] create auto-comment workflow

LongLiveCHIEF, 2021-10-17:
FYI... this is super easy to accomplish with the https://github.com/actions/github-script#comment-on-an-issue action. If we go python, we could probably package it as a composite action and have a template file in that composite action folder and then have the python script load that up.

Although, iirc we'd have to juggle secrets at that point so the action would have the ability to use the github sdk pypi library to interact with the PR in this fashion.

LongLiveCHIEF, 2021-10-17:
T1 boards would only get tested on a change to the HAL or their associate PAC.  I've also found that it is great to set a clear expectation of what the next step is, and whose responsibility that step is, when a PR fails.

It seems redundant, especially if we call these things out in contributor documentation, but it can make things go _way_ smoother.  It's one of those "nice touches" that really helps grease the wheels of a project and keep things running smoothly.

bradleyharden, 2021-10-17:
Ok, yeah that makes sense. It will definitely help with new contributors.