atsamd-rs/atsamd #566: atsamd-rs:owners should own crates published by atsamd-bot

sajattack, 2022-01-11:
@TDHolmes noticed that some of the crates recently published through github actions were owned by only atsamd-bot, and not atsamd-rs:owners. We should update the github action script (atsamd/.github/workflows/release-crates.yml) to include a command such as `cargo owner add github:atsamd-rs:owners` (not sure if this is the exact syntax, please double check)

jbeaurivage, 2024-09-06:
Having looked through [here](https://crates.io/teams/github:atsamd-rs:owners), I can confirm that the `atsamd-rs` team members are owners of all crates we currently maintain. Not sure if we should try to add the team members as owners every time a crate is published, or as needed as new crates are created.

ianrrees, 2024-09-07:
Cool, so I guess after we have #757 merged, we can see what happens when we release a crate then close this if it is owned by the atsamd-rs team?

jbeaurivage, 2024-10-19:
Can confirm that all crates released successfully.

closed 2024-10-19 by jbeaurivage