atsamd-rs/atsamd #582: Support L Variant SAMD[2|A]1E Parts

kiranshila, 2022-03-18:
As far as I can tell - SAMD[2|A]1ExL parts are not supported by this crate - is that correct? Specifically, this is the variant with a pinout optimized for ADC/PWM usage. I am using PB02 and PB03 in a design, but am getting

```console
error[E0609]: no field `pb03` on type `atsamd_hal::gpio::v2::Pins`
  --> src/bsp.rs:10:1
   |
10 | / hal::bsp_pins!(
11 | |     // GPIO
12 | |     PA07 { name: rf1_stat_led },
13 | |     PA06 { name: rf2_stat_led },
...  |
43 | |     }, //ALTC_SERCOM0[3]
44 | | );
   | |_^ help: a field with a similar name exists: `pa03`
```

when trying to use `bsp_pins!` on a SAMD21E17L part. I'm not sure if this is as simple as expanding the features in https://github.com/atsamd-rs/atsamd/blob/master/hal/src/gpio/v2/pin.rs#L1124 or if there are more implications.

kiranshila, 2022-03-18:
Here is the app note that talks about the differences

kiranshila, 2022-03-18:
I'm working on a PR here #583

sajattack, 2022-03-20:
I noticed these chips have separate SVDs in newer atpack files from https://packs.download.microchip.com/ , but whether we need to generate entire PACs, I am not sure.

bradleyharden, 2022-03-20:
@sajattack, we could probably just diff the SVDs. Isn't that how we decided to get rid of the memory-specifier fields of the part number?

TDHolmes, 2022-04-01:
I believe @kiranshila diffed them and there wasn't anything important. Is that right?

TDHolmes, 2022-04-01:
Closing now that #583 merged

closed 2022-04-01 by TDHolmes