atsamd-rs/atsamd #594: [arduino_mkrzero] Trying to use builtin SD card reader: TimeoutWaitNotBusy and unexpected reset

arkap, 2022-03-30:
I'm currently trying to get the arduino MKRZERO's builtin SD card reader to work. I've made sure my hardware works by uploading and running an arduino sketch and I've used the `feather_m0`'s `adalogger` example as a starting point. This is how I configure the SPI
```rust
    let timer_clock = clocks
        .configure_gclk_divider_and_source(ClockGenId::GCLK3, 32, ClockSource::OSC32K, true)
        .unwrap();
    let rtc_clock = clocks.rtc(&timer_clock).unwrap();
    let timer = rtc::Rtc::clock_mode(peripherals.RTC, rtc_clock.freq(), &mut peripherals.PM);
    let gclk0 = clocks.gclk0();
    let spi_clock = clocks.sercom4_core(&gclk0).unwrap();
    let sd_spi_pads = spi::Pads::<Sercom4>::default()
                .sclk(pins.pa13)
                .data_in(pins.pa15)
                .data_out(pins.pa12);
    let spi_config = spi::Config::new(
            &peripherals.PM,
            peripherals.SERCOM4,
            sd_spi_pads,
            spi_clock.freq()
        );
    let spi = spi_config
            .baud(400.khz())
            .spi_mode(hal::ehal::spi::MODE_0)
            .enable();

    let sd_cd = pins.pa27.into_pull_up_input();
    let mut sd_cs = pins.pa14.into_push_pull_output();
```
I noticed that the logic of the chip detect pin `sd_cd` seems to be inverted. So I'm waiting until `sd_cd` is low and try to initialize:
```rust
    if sd_cd.is_high().unwrap() {
        usbserial_write!("No card detected. Waiting...\r\n");
        while sd_cd.is_high().unwrap() {
            delay.delay_ms(250_u32);
        }
    }
    usbserial_write!("Card inserted!\r\n");

    let mut controller = Controller::new(SdMmcSpi::new(spi, sd_cs), timer);
    controller.device().init(); // <-- this results in TimeoutWaitNotBusy after some time
```
What's weird is that I run into the timeout only if the card is already inserted when I power on the MKRZERO. Plugging in a card while it is running seems to reset the device. I'd be grateful for any advise or pointers to helpful resources.

TDHolmes, 2022-04-01:
I made the adalogger example and it worked for me back then.. Are you pulling `sd_cs` high after initializing it? I'm also wondering if lowering the SPI speed even lower helps, or maybe some delay between detecting chip inserted and trying to initialize helps.

Can you tell what kind of reset occurs and why? Could you be panicking?

arkap, 2022-04-20:
Thanks for the ideas. I do pull `sd_cs` high. I accidentally clipped that line from my snippet above initially. A lower speed hasn't helped, unfortunately.
How can I find out which kind of reset occurs? I'm using `panic_halt` so I expect the chip to stop in case of a panic and not reset.