atsamd-rs/atsamd #599: bsp_pins! macro error when no pin definitions include aliases

datagrok, 2022-04-01:
Attempting to create a simple BSP for a custom PCB and had a problem using the `bsp_pins!` macro. When I omit `aliases: { ... }` from **all** my pin definitions, rust complains about "missing tokens in macro arguments." Adding an `aliases: { ... }` spec to any one or more pins clears the compiler error.

This code fails to compile:
```rust
...
pub mod pins {
    use super::hal;
    hal::bsp_pins!(
        PB12 { name: sca }
        PB13 { name: scl }
        }
    );
}
```
Failing with,
```
error: unexpected end of macro invocation
  --> src/lib.rs:13:5
   |
13 | /     hal::bsp_pins!(
14 | |         PB12 { name: sca }
15 | |         PB13 { name: scl }
16 | |     );
   | |_____^ missing tokens in macro arguments
   |
   = note: this error originates in the macro `$crate::__define_pin_alias_macro` (in Nightly builds, run with -Z macro-backtrace for more info)
```

This code, however, compiles:

```rust
...
pub mod pins {
    use super::hal;
    hal::bsp_pins!(
        PB12 { name: sca }
        PB13 {
            name: scl
            aliases: {
                AlternateC: SCL
            }
        }
    );
}
```

Tested with **atsamd-hal 0.15.0** and **atsamd21j 0.12.0**.
**Rust stable 1.59.0**, but same error occurs in nightly.

I'm able to proceed using that workaround but it was awfully confusing at first.

TDHolmes, 2022-04-01:
I **_think_**
```rust
                   aliases: {
                        $(
                            $( #[$alias_cfg:meta] )*
                            $Mode:ident: $Alias:ident $(,)?
                        )+
                    }
```

needs to be changed to
```rust
                   aliases: {
                        $(
                            $( #[$alias_cfg:meta] )*
                            $Mode:ident: $Alias:ident $(,)?
                        )*
                    }
```

bradleyharden, 2022-04-01:
There has been a ton of activity with the HAL lately, and I've been slammed at work, but I will try to look at this soon. Sorry for the delay

datagrok, 2022-04-02:
I imagine it's possible that zero pins having aliases isn't a real-world use case, and if so, there's no reason to complicate the macro definition.

My initial encounter with this problem was a minimalist attempt to see if I was using the macro correctly. "Will it compile with a single named pin? If so then I'll go ahead and define the rest of my pins, and then figure out what aliases I should add."

Regardless, thanks for taking a look!

TDHolmes, 2022-04-02:
At a minimum, documentation should be updated

bradleyharden, 2022-04-24:
@datagrok, please give #605 a try. I did a quick test on my end. I was able to reproduce the problem, and my change fixes it, so I think it should be good.

bradleyharden, 2022-05-22:
@datagrok, did you have a chance to try this?

PR #605 (2022-06-10): Fix failing `bsp_pins!` invocation with no aliases
When no aliases are included in a call of `bsp_pins!`, the resulting
`pin_alias!` macro expands to a completely empty macro definition, which
is invalid.

Add identity "aliases" for each given pin name. This has two benefits.
First, it prevents the empty macro case above. And second, it precludes
any possibility of collisions between names and aliases, since all names
are now checked before resolving aliases.

Closes #599

# Checklist
  - [x] `CHANGELOG.md` for the BSP or HAL updated
  - [x] All new or modified code is well documented, especia [... cut]
commit 2dad8315 (2022-06-10): Fix failing `bsp_pins!` invocation with no aliases (#605)

* Fix failing `bsp_pins!` invocation with no aliases

When no aliases are included in a call of `bsp_pins!`, the resulting
`pin_alias!` macro expands to a completely empty macro definition, which
is invalid.

Add identity "aliases" for each given pin name. This has two benefits.
First, it prevents the empty macro case above. And second, it precludes
any possibility of collisions between names and aliases, since all names
are now checked before resolving aliases.

Closes #599

* Update CHANGELOG

* Fix CHANGELOG
closed 2022-06-10 by jbeaurivage