atsamd-rs/atsamd #603: Recent version bump broke essentially all text rendering

cbiffle, 2022-04-13:
PR #591 bumped versions of dependencies. While the PR description described the bump as "minor," many of the crates that were bumped are in the leading-0 phase of semver, meaning that minor version bumps are breaking.

One good example of breakage is the `embedded_graphics` crate, which appears to have switched the default baseline behavior for text rendering in (I'm guessing) 0.7. As a result, after #591, many of the examples render their text off the screen, and programs that attempt to clear the screen before writing text (the `clock` examples, say) instead clear a rectangle just beneath the text.

Fixing this generally requires diffs like the following:

```diff
diff --git a/boards/wio_terminal/examples/clock.rs b/boards/wio_terminal/examples/clock.rs
index 9e4567f29..2f0c02f72 100644
--- a/boards/wio_terminal/examples/clock.rs
+++ b/boards/wio_terminal/examples/clock.rs
@@ -15,7 +15,7 @@ use eg::mono_font::{ascii::FONT_10X20, MonoTextStyle};
 use eg::pixelcolor::Rgb565;
 use eg::prelude::*;
 use eg::primitives::{PrimitiveStyleBuilder, Rectangle};
-use eg::text::Text;
+use eg::text::{Baseline, Text};
 
 use cortex_m::interrupt::free as disable_interrupts;
 use cortex_m::peripheral::NVIC;
@@ -137,7 +137,7 @@ fn main() -> ! {
             .ok()
             .unwrap();
 
-        Text::new(data.as_str(), Point::new(55, 80), style)
+        Text::with_baseline(data.as_str(), Point::new(55, 80), style, Baseline::Top)
             .draw(&mut display)
             .ok()
             .unwrap();
```

Anyway, noticed this when I was using your code to get a shiny new Wio Terminal working before I write my own drivers. :-)

TDHolmes, 2022-04-13:
If you have the change to fix, please open a PR! Thanks for the report

cbiffle, 2022-04-13:
I have the fix for the one example I used to test flashing (above); I'm not equipped to test all the examples. I also can't promise this is the only breaking change in embedded-graphics 0.7.

TDHolmes, 2022-04-13:
As you can tell by us breaking the examples, we don't necessarily have all of the hardware on hand at all times either 🙂 even a partial fix PR is appreciated

bradleyharden, 2022-04-24:
> As you can tell by us breaking the examples, we don't necessarily have all of the hardware on hand at all times either slightly_smiling_face even a partial fix PR is appreciated

[Evidence of, and apology for, our poor behavior](https://github.com/atsamd-rs/atsamd/pull/542#issuecomment-1000926843)

pabigot, 2022-12-04:
@bradleyharden Between the merge of #604 and my pending patches I believe all text rendering in examples should be resolved.

bradleyharden, 2022-12-26:
@pabigot, have all the relevant patches been merged? I think so. If so, we can close. @cbiffle, can you confirm your issues are resolved?

pabigot, 2022-12-27:
@bradleyharden All my patches are in and the examples work for me on the current master branch.

bradleyharden, 2022-12-27:
Ok, I'm going to close then. @cbiffle, feel free to reopen if you're still having an issue.

closed 2022-12-27 by bradleyharden