atsamd-rs/atsamd #609: Initializing SPI randomly hangs

sakian, 2022-05-13:
Hi, I'm seeing random hangs when I initialize the SPI bus. It seems to happen ~10% of the time when running from debug, and a bit less frequent when running from release. It's hanging on [this line](https://github.com/atsamd-rs/atsamd/blob/54cde4c7fd19ac3217ef1093a85aec1032736149/hal/src/sercom/spi/reg.rs#L108).

My code:
```
spi::Config::new(mclk, sercom3, pads, clock.freq())
        .baud(100.mhz()) // Max speed for ADC
        .length::<spi::lengths::U2>() // 16 bit
        .bit_order(spi::BitOrder::MsbFirst)
        .spi_mode(spi::MODE_0) // CPOL = 0, CPHA = 0
        .enable()
```
and here's the stack backtrace after I halt the app when hung:
```
stack backtrace:
   0: atsamd_hal::sercom::spi::reg::Registers<S>::set_length
        at C:\Development\Vital\device-sw\deps\atsamd\hal\src\sercom\spi\reg.rs:108:15
   1: atsamd_hal::sercom::spi::Config<P>::default
        at C:\Development\Vital\device-sw\deps\atsamd\hal\src\sercom\spi.rs:660:9
   2: atsamd_hal::sercom::spi::Config<P>::new
        at C:\Development\Vital\device-sw\deps\atsamd\hal\src\sercom\spi.rs:697:9
   3: bsp::detector::init_spi
        at C:\Development\Vital\device-sw\phoenix-firmware\bsp\src\detector.rs:121:5
   4: app::app::init
        at src\main.rs:228:28
   5: app::app::rtic_ext::main::{{closure}}
        at src\main.rs:11:1
   6: app::app::rtic_ext::main::__rtic_init_resources
        at src\main.rs:11:1
   7: cortex_m::asm::nop
        at C:\Users\MarkBremer\.cargo\registry\src\github.com-1ecc6299db9ec823\cortex-m-0.7.4\src\asm.rs:35:5
   8: main
        at src\main.rs:11:1
   9: Reset
```

sakian, 2022-05-13:
Obviously, if I don't use the 32-bit extension it works ok, but the HAL defaults to using it

bradleyharden, 2022-05-13:
@sakian, my first guess would be that the order of initialization matters. Check the datasheet and see what it says on initialization order. I know I ran into that issue once before.

If that turns out to be the cause, then we should probably find a way to handle initialization order for users, so you don't need to comb through the datasheet just to set up a SPI bus.

If that's not the cause, then I'm not sure. I'll have to dig more

sakian, 2022-05-13:
Thanks, @bradleyharden, I've rearranged the init order to match what they're looking for and will try it out.

bradleyharden, 2022-05-22:
@sakian, any luck?

sakian, 2022-05-22:
Unfortunately, I've been tied up on other things. I tried reverting my changes and replicating the issue, but it's quite transient so it didn't happen again. I'll have to wait till it shows up again and reorganize the function call order. It does make sense that it's the issue though because I didn't follow the order specified in the datasheet.

Mark Bremer

sakian, 2022-05-26:
I saw the issue again, and after reorganizing the commands to match the order defined in the datasheet (section 35.6.2.1) the issue went away. It's a very random issue, so I'm not sure if the change actually fixed it, but I'll keep this change in place to see if it happens again.