atsamd-rs/atsamd #619: No way to set pull-down or pull-up for alternate functions

sakian, 2022-06-07:
I'm looking to enable the pull-up for the RX pin of a UART, but it doesn't look like this is possible. The `change_mode` function within `gpio/reg.rs` sets the pullen and out fields, but sets them to disabled for all alternate functions. Is there a way I can force these to enable the pull-up functionality for a specific pin?

bradley-harden-apl, 2022-06-07:
@sakian, I think we confirmed that you can't actually use pull resistors with SERCOM functions. If I remember correctly, the datasheet implies that you can, but it doesn't work when tested. However, it *does* work for external interrupts, which is why we changed `AlternateA` to `Interrupt` with corresponding pull states.

@jbeaurivage, can you confirm all this? I don't remember where that discussion would have occurred. Maybe on Matrix?

bradleyharden, 2022-06-07:
@sakian, I think we confirmed that you can't actually use pull resistors with SERCOM functions. If I remember correctly, the datasheet implies that you can, but it doesn't work when tested. However, it *does* work for external interrupts, which is why we changed `AlternateA` to `Interrupt` with corresponding pull states.

@jbeaurivage, can you confirm all this? I don't remember where that discussion would have occurred. Maybe on Matrix?

sakian, 2022-06-07:
That's what I thought, but the electronics guy I work with wants me to try it. Maybe there's a different way to do it. Is there a way to set those registers through the PAC after the HAL has configured it?

bradleyharden, 2022-06-07:
You could steal another copy of the `Peripherals` struct and use that. I don't remember it exactly, but it would be something like

```rust
let p = unsafe { Peripherals::steal() };
p.PORT.pincfg[id].write(...);
```

bradleyharden, 2022-06-08:
Maybe it was @ianrees I was thinking of?

closed 2022-06-10 by sakian