atsamd-rs/atsamd #623: Is the ATSAMDA1 micro supported?

sakian, 2022-06-24:
I'm looking to develop for a ATSAMDA1G16B microcontroller. It isn't listed in this repo as a supported chip, but it shares a datasheet with the D21 chips. Is it supported out-of-the-box? Do I need to add support to get it to work properly (add PAC, add to HAL)?

sajattack, 2022-06-24:
I'm not sure if that chip existed when this project began. :sweat_smile: Anyway, I would recommend looking through the datasheet to see how it compares to other chips we support in order to determine if it needs a separate pac or can use one of the existing ones.

sakian, 2022-06-24:
From diffing ATSAMD21G16B.svd with ATSAMDA1G16B.svd, it looks like there are only a few differences:
- The DSU peripheral has a different DID reset vals (makes sens since it's a different chip model) and the entry fields are enumerated differently
- The DA1 has one less GCLK (8) opposed to the D21 (9)
- The DA1 has two more TCs (TC6 and TC7)

Actually, the ATSAMD21J16B is an even closer match since it also has the extra TCs, although interestingly, it looks like the HAL doesn't support them.

Does anyone anticipate any issues going ahead with treating the DA1G chip as a D21J?

Dridus, 2022-08-30:
FWIW, we (Vital) added a proper PAC and HAL adjustments based on the SVD, though we're still developing against it so might find issues. it's also based on the WIP v2 clocking branch. if anybody needs ATSAMDA1 support though they might find it useful: https://github.com/VitalBio/atsamd/pull/1

ianrrees, 2025-03-11:
Closing as this looks stale, can reopen if required down the line.

closed 2025-03-11 by ianrrees