atsamd-rs/atsamd #628: wio-terminal WiFi examples broken by #542

pabigot, 2022-08-23:
As noted in https://github.com/atsamd-rs/atsamd/pull/604#issuecomment-1099736323 the Wi-Fi examples for this target don't work; bisect narrows it down to 34ad122ab2cb9510d69fa465c1f83c3eda59c092.  Observable behavior is that as soon as interrupts are re-enabled after wifi,enable() everything goes dark.

Between the board not supporting debugging without hardware modifications, and Rust blocking my normal dirty tricks to toggle LEDs without regard to safety, I don't expect to be able to solve this, but if somebody has suggestions I can test them.

lokiodin, 2022-09-26:
Hello @pabigot,
I encounter the same thing. 
Did you succeed to make it work ?

pabigot, 2022-09-26:
@lokiodin I haven't had a chance to get back to this yet.

lokiodin, 2022-09-26:
@pabigot Ok thanks. I will try to do something but as I a new in the embedded world ...

garasubo, 2022-12-04:
I encountered the same bug. I used the debugger, and found that the reason was `SERCOM0_2` interrupt handler was crashed due to `FrameError` in `Wifi::_handle_rx` (https://github.com/atsamd-rs/atsamd/blob/18cf7e9bbae1df869f4d39e8b67b401a4120e66e/boards/wio_terminal/src/wifi.rs#L172).
But I'm not sure why this error happened since I don't know much knowledge about how this module works.

pabigot, 2022-12-04:
@garasubo Awesome; thanks!  That inspired me to try again, and I found a fix that's in #644.

Could you point me to any information you can cite or provide on how to use a debugger on these boards?  Ideally without having to modify the hardware to enable SWD, but if that's necessary I might be able to find somebody to do it.

closed 2022-12-04 by bradleyharden

garasubo, 2022-12-05:
The SWD connector is available inside Wio Terminal case. I connected my Wio Terminal to Seeduino Xiao using FPC cable.
Here is the detail (JP): https://github.com/tomoyuki-nakabayashi/Embedded-Rust-from-Basics/tree/main/debug

You also need DAPLink for Xiao and openocd from here: https://github.com/ciniml/debug-tools-builder/releases

garasubo, 2022-12-05:
@pabigot Hmm, I tried your fix, but it didn't work for me. I tried `wifi_scan` example in the latest master (2aef0d18a78a4219e2a74f194949517b25f88262), and it still got panic in the same place.

garasubo, 2022-12-07:
> There are a couple things that might be going wrong. First, the wifi_scan sample requires the patch from https://github.com/atsamd-rs/atsamd/pull/647 otherwise the display doesn't work.

That was the reason. Thanks.
I've already updated the firmware to the latest, and tested with official Arduino libraries. So, my hardware and firmware were fine. After switching #647 branch, it started to show wifi information on the display. Thanks. Hope #647 get merged and the new version is published with these change soon!

garasubo, 2022-12-11:
Hmm, it seems still broken in debug build. Is it better to open a new issue for it?

pabigot, 2022-12-11:
I think so, yes.

pabigot, 2022-12-19:
@garasubo I don't think this is a bug.  The connection to the RTL8720 has no flow control, so the Overflow error that causes a panic is legitimate.  The example works if `opt-level = 1` is added to `profile.dev`.

PR #644 (2022-12-04): Fix wio-terminal Wi-Fi
# Summary
Delay enable of the UART until the device has been reset.

Fixes #628