atsamd-rs/atsamd #636: I2C slave mode support

tgross35, 2022-11-03:
What would be needed to support i2c slave mode? It seems like there is a lot of overlap with i2c master, but I am not sure what the structure would look like.

Register summary is in section 36.7 of the datasheet

sajattack, 2022-11-04:
I think the main reason we don't have it yet is mainly a lack of standardization for the interface in the embedded-hal crate (last I checked). But that's not exactly a good reason why we can't or shouldn't support it.

tgross35, 2022-11-04:
Now that it looks like embedded-hal-bus is going to be split off, I wonder if it will cover it better. The details about what HAL 1.0 are going to look like are unfortunately iffy at best it seems.

Err... I'm pretty new to the project, could you point me in the right direction for where to start adding something like this?

sajattack, 2022-11-04:
If I were to do it, I would create a `slave` module under the existing i2c directory https://github.com/atsamd-rs/atsamd/tree/master/hal/src/sercom/i2c

sajattack, 2022-11-04:
We also have a chatroom where we can discuss if you have more questions https://matrix.to/#/#atsamd-rs:matrix.org