atsamd-rs/atsamd #649: How can I use usb function on 0.15.0 or higher

N-Magi, 2022-12-19:
I am working on a project using the usb function on the Arduno MKR ZERO.
I was running it with some sample code, but the usb function did not work when the library version was 0.15.0 or higher.
The same code worked fine when using 0.14.0.

How can we use the usb function properly with 0.15.0 or higher?

Show in brew is code
```

#![no_main]
#![no_std]

use atsamd_hal as hal;
use cortex_m_rt::entry;
use hal::pac;

use hal::gpio::AnyPin;
use hal::gpio::*;

use hal::prelude::*;

use atsamd_hal::clock::GenericClockController;
use atsamd_hal::delay::Delay;

use atsamd_hal::gpio::Pins;
use atsamd_hal::pac::interrupt;
use atsamd_hal::pac::{CorePeripherals, Peripherals, NVIC};
use atsamd_hal::usb::usb_device::bus::UsbBusAllocator;
use atsamd_hal::usb::UsbBus;
use usb_device::prelude::*;
use usbd_serial::{SerialPort, USB_CLASS_CDC};

extern crate panic_halt;

static mut USB_ALLOCATOR: Option<UsbBusAllocator<UsbBus>> = None;
static mut USB_BUS: Option<UsbDevice<UsbBus>> = None;
static mut USB_SERIAL: Option<SerialPort<UsbBus>> = None;
// static INTERRUPT_FIRED: atomic::AtomicBool = atomic::AtomicBool::new(false);

pub fn usb_allocator(
    usb: pac::USB,
    clocks: &mut GenericClockController,
    pm: &mut pac::PM,
    dm: impl AnyPin<Id = PA24>,
    dp: impl AnyPin<Id = PA25>,
) -> UsbBusAllocator<UsbBus> {
    let gclk0 = clocks.gclk0();
    let usb_clock = &clocks.usb(&gclk0).unwrap();

    UsbBusAllocator::new(UsbBus::new(usb_clock, pm, dm, dp, usb))
}

#[entry]
fn main() -> ! {
    let mut peripheral = Peripherals::take().unwrap();
    let mut core = CorePeripherals::take().unwrap();
    let mut clks = GenericClockController::with_external_32kosc(
        peripheral.GCLK,
        &mut peripheral.PM,
        &mut peripheral.SYSCTRL,
        &mut peripheral.NVMCTRL,
    );
    let mut pins = Pins::new(peripheral.PORT);

    let bus_allocator = unsafe {
        let dm = pins.pa24;
        let dp = pins.pa25;
        //let (dm, dp) = (dm.into(), dp.into());
        USB_ALLOCATOR = Some(usb_allocator(
            peripheral.USB,
            &mut clks,
            &mut peripheral.PM,
            dm,
            dp,
        ));
        USB_ALLOCATOR.as_ref().unwrap()
    };

    unsafe {
        USB_SERIAL = Some(SerialPort::new(bus_allocator));
        USB_BUS = Some(
            UsbDeviceBuilder::new(bus_allocator, UsbVidPid(0x2222, 0x3333))
                .manufacturer("Fake")
                .product("Serial port")
                .serial_number("TEST")
                .device_class(USB_CLASS_CDC)
                .build(),
        );
    }

    unsafe {
        core.NVIC.set_priority(interrupt::USB, 1);
        NVIC::unmask(interrupt::USB);
    }

    let mut delay = Delay::new(core.SYST, &mut clks);
    let mut led = pins.pb08.into_push_pull_output();

    loop {
        delay.delay_ms(500u32);
        led.set_high().unwrap();
        delay.delay_ms(500u32);
        led.set_low().unwrap();

    }
}

fn poll_usb() {
    unsafe {
        USB_BUS.as_mut().map(|usb_dev| {
            USB_SERIAL.as_mut().map(|serial| {
                usb_dev.poll(&mut [serial]);

                // Make the other side happy
                let mut buf = [0u8; 16];
                let _ = serial.read(&mut buf);
            });
        });
    };
}

#[interrupt]
fn USB() {
    poll_usb();
}

```

sajattack, 2022-12-19:
The gpio module was replaced in 0.15.0. Refer to docs at https://docs.rs/atsamd-hal/0.15.1/atsamd_hal/gpio/index.html and similar samples for tier1-supported boards (which have been updated to use gpio-v2 in their examples)

ianrrees, 2024-01-18:
I suspect the issue here was to do with a debug build; in the 48MHz micros these wind up being too slow to reliably enumerate.

I'm going to close this as it's been idle for quite a long time, but if the problems persist with release builds of current HALs, feel free to reopen it!

closed 2024-01-18 by ianrrees