atsamd-rs/atsamd #660: Correct alternate for USB pins on some M4 BSPs?

jboynes, 2023-01-05:
In the [feather_m4](https://github.com/atsamd-rs/atsamd/blob/master/boards/feather_m4/src/lib.rs#L154) and [metro_m4](https://github.com/atsamd-rs/atsamd/blob/master/boards/metro_m4/src/lib.rs#L218) BSPs, pins PA24 and PA25 define the USB function as `AlternateG`. However, in the D51 datasheet this is defined as mux `H`. It _is_ `G` on D21 parts.

The [atsame54_xpro](https://github.com/atsamd-rs/atsamd/blob/master/boards/atsame54_xpro/src/pins.rs#L166) defines USB as `AlternateH`.

I don't have these boards to verify on but want to raise it as a potential issue.

bradleyharden, 2023-01-05:
@jboynes, good catch. I think that was just a copy/paste error from the feather_m0 crate. It looks like the [HAL is correct](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/thumbv7em/usb/mod.rs#L16-L23), so it should only result in a type error, not code that runs but doesn't work. Could you please submit a PR?

jboynes, 2023-01-05:
PR #661 created for this.

Is there another issue here in that this _should_ result in code that either doesn't compile or doesn't work (specifically, appears to work but doesn't actually interact with the physical bus)?

bradleyharden, 2023-01-05:
Ha, it looks like it's actually the opposite. Instead of silently failing, it silently corrects the problem.

The BSP [`usb_allocator`](https://github.com/atsamd-rs/atsamd/blob/master/boards/feather_m4/src/lib.rs#L263-L277) function converts the pins to `AlternateG` and then passes them to `UsbBus::new`. But [`UsbBus::new`](https://github.com/atsamd-rs/atsamd/blob/master/hal/src/thumbv7em/usb/bus.rs#L517-L542) doesn't actually care which `PinMode` is already configured. Instead, it always converts it to `AlternateH` itself.

Really, that means the `bsp_allocator` function is over-specifying its inputs. It only needs to require pins `PA24` and `PA25`, not the correct `PinMode`. But doing that requires the `AnyPin` trait and makes the signature slightly more complicated, so it's easier to just constrain it to `impl Into<UsbDp>`, etc.

bradleyharden, 2023-01-15:
Closed with #661

closed 2023-01-15 by bradleyharden