atsamd-rs/atsamd #680: atsamd21e18a I2C

hugolgst, 2023-03-31:
Hello,

I am writing code for this [board](https://www.microchip.com/en-us/development-tool/DM320118) using ATSAMD21E18A.
I would like to initiate an I2C connection but am failing with this code: 
[image]
following https://docs.rs/atsamd-hal/latest/atsamd_hal/sercom/i2c/index.html 

The LED stays on and nothing is happening, I am not sure why

sakian, 2023-03-31:
Doesn't look like you ever control the LED? Can try adding `pa02.set_high().unwrap()` or `pa02.set_low().unwrap()`?

sakian, 2023-03-31:
I think you need to configure the peripheral clock then. Check this example out: https://github.com/atsamd-rs/atsamd/blob/e8fbf5de0228b19528f8279613149d9601eabbda/boards/metro_m0/examples/i2c.rs#L56

When you enable, it waits for the register to be synchronized, which won't happen if the sercom isn't given a clock. You'll need to pass enable the peripheral clock, like in the example I linked.

As an aside, it would be nice to have the `Config::new` call take the whole clock, instead of just the clock frequency to make this impossible from happening.

hugolgst, 2023-04-03:
@sakian  Thanks, that was it. Since we are here I have another question, I am trying to communicate with this chip https://ww1.microchip.com/downloads/aemDocuments/documents/SCBU/ProductDocuments/DataSheets/ATECC608B-TNGTLS-CryptoAuthentication-Data-Sheet-DS40002250A.pdf and have successfully executed the wake call with the proper response but for some reason am getting a NACK with a payload that I know is working (tested and sent in another programming language). I suspect that the write function has a different behaviour than the one in Go (tinygo), but could be wrong. 
[image]

would you have an idea why I am getting a NACK in the two instances of i2c.write?

sakian, 2023-04-04:
I don't see anything wrong that sticks out to me. Probably need to go through standard debugging (make sure port is configured properly to talk to that chip, confirm data being sent is correct, probe data on wires to make sure it's correct, etc.).

ianrrees, 2023-11-14:
@hugolgst - did you get this sorted?

Gentle suggestion: rather than posting images of code, it's helpful to use code blocks so that others can easily copy-and-paste, use search, etc.  For bigger pieces of code, a [GitHub gist](https://gist.github.com/) is be a good option too.

closed 2023-11-23 by ianrrees