atsamd-rs/atsamd #682: Failed to erase flash sector at address 0x00000000 Feather_m0

nvsd, 2023-04-28:
**Describe the bug**
I'm unable to use cargo-flash, probe-run, and cargo-embed due to the below error while using a Jlink EDU Mini with a feather M0 proto
```
Caused by:
    0: Failed to erase flash sector at address 0x00000000.
    1: The execution of 'erase_sector' failed with code 1. This might indicate a problem with the flash algorithm.
```

**To Reproduce**
Steps to reproduce the behavior:

1. Ensure Jlink and feather board are plugged in 
2. Clone [atsamd-rs board example](https://github.com/atsamd-rs/atsamd/tree/master/boards/feather_m0)
3. Run cargo run
4. observe the error

**Expected behavior**
Able to flash and debug device

**Stacktrace**
If applicable, add a stacktrace to help explain your problem.

**Desktop (please complete the following information):**

- macOS Venture (13.2.1 (22D68)) m1

**Additional context**
I can flash the board using hf2 and bossac so the board works. I can also connect to the board using Jlink commander.
I assume I have a configuration issue, but I'm unsure. Let me know what else I can provide.

`cargo/config`
```
# samd21 is a Cortex-M0 and thus thumbv6m
[build]
target = "thumbv6m-none-eabi"
rustflags = [
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  "-C", "link-arg=--nmagic",
  "-C", "link-arg=-Tlink.x",
]

[target.thumbv6m-none-eabi]
# runner = 'arm-none-eabi-gdb'
runner = 'probe-run --chip ATSAMD21G18A'
```
`Embed.toml`
```
[default.general]
chip = "ATSAMD21G18A"

[default.rtt]
enabled = true
```

`memory.x`
```
MEMORY
{
  /* Leave 8k for the default bootloader on the Feather M0 */
  FLASH (rx) : ORIGIN = 0x00000000 + 8K, LENGTH = 256K - 8K
  RAM (xrw)  : ORIGIN = 0x20000000, LENGTH = 32K
}
_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```

nvsd, 2023-04-29:
Looks like its a probe-rs issue
https://github.com/probe-rs/probe-rs/issues/1599

closed 2023-04-29 by nvsd

sajattack, 2023-04-29:
This currently only supports ATSAME51, but it sounds like you need something similar for atsamd21 if I understand correctly. https://github.com/atsamd-rs/atsamd-flash-algorithm