atsamd-rs/atsamd #699: Incorrect values in SVD files

johngigantic, 2023-09-30:
Probe-rs threw some errors when opening debug configurations with the ATSAME54P20A SVD file in this repository. The following lines are in question - there is a single hex digit typo in three resetValue fields.

For the ATSAME54P20A SVD file, the errors are as follows:

Line 27591: the resetValue should be 0x00, not 0xE00.
`
            <register>
               <name>HC1R</name>
               <description>Host Control 1</description>
               <addressOffset>0x28</addressOffset>
               <size>8</size>
               <resetValue>0x00</resetValue>
`

Line 27720: the resetValue should be 0x00, not 0xE00, as in the last error.
`
            <register>
               <name>HC1R_EMMC_MODE</name>
               <description>Host Control 1</description>
               <alternateRegister>HC1R</alternateRegister>
               <addressOffset>0x28</addressOffset>
               <size>8</size>
               <resetValue>0x00</resetValue>
`

Line 31929: the resetValue should be 0x0000, not 0x20000.
`
               <register>
               <name>SISR</name>
               <description>Slot Interrupt Status</description>
               <addressOffset>0xFC</addressOffset>
               <size>16</size>
               <access>read-only</access>
               <resetValue>0x20000</resetValue>
`

johngigantic, 2023-09-30:
More to the point, errors such as these may be present on several of the SVD files. Probe-rs was able to detect these ones in particular because the values did not fit in the bitfield size.

Are there SVD files which are more up-to-date with these errors fixed? Is it worth just trying to fix all these values manually - which may not be too bad if probe-rs detects them and there's only 1-5 per SVD file?

sajattack, 2023-10-02:
The svd files come from Microchip atpacks. We can patch or update them if a better version is available. If a patch is needed see here https://github.com/atsamd-rs/atsamd/blob/master/svd/devices/include/atsamd5x.xsl

pustekuchen91, 2023-12-04:
I already openend two tickets @ microchip for this issue. They are aware of the problem and want to fix this in the future. But they cant tell when. There are serveral more svd files with the same error. e.g. ATSAME53J20, ATSAME53N20

jbeaurivage, 2024-10-21:
@johngigantic, those fixes are already included in our SVD patches, which get applied every time we rebuild PACs. See here: https://github.com/atsamd-rs/atsamd/blob/f790fad0d51832b874ff07531e99ccab3bb5245b/svd/devices/include/atsamd5x.xsl#L9

Note that this patch is applied to all SVDs for the SAMx5x devices we support.

closed 2024-10-21 by jbeaurivage