atsamd-rs/atsamd #700: changing manufacturer in the UsbDeviceBuilder

charleselie, 2023-10-16:
Hello,

I am building an app based on rust with a GUI in slint. 
This app create dialog with a feather board (or equivalent).

I can't see the change of manufacturer with feather M4 and itsy bitsy on my app on my computer.

I've used serialPort for my app on my computer, and the UsbDeviceBuilder on my feather.
The manufacturer info is still showing "Microsoft", but the serial number  is changing depending on the value I use on the feather.

is this normal ?

sajattack, 2023-10-16:
It sounds like maybe it's using the microsoft usb-serial driver and showing that? But I'm only speculating

charleselie, 2023-10-17:
It seems to be a problem based on the Serialport library, that doesn't get the correct information.

ianrrees, 2023-11-15:
If there's an issue with the way the manufacturer information is supplied over USB, it will probably be in either [usbd-serial](https://github.com/rust-embedded-community/usbd-serial) or [usb-device](https://github.com/rust-embedded-community/usb-device) which are used by our example code.  However, then it's up to the host computer to use the manufacturer info appropriately, and it sounds like in this case the Microsoft driver or the higher-level application might not be doing what is desired.

@charleselie - have you figured out the problem?  If not, I'm happy to help diagnose as I'm a bit involved in usb-device, but you'll probably need to help with a minimal reproduction and doing some testing since I don't have a real Windows computer.