atsamd-rs/atsamd #706: ATSAME53N20A ADC question

ianrrees, 2023-12-04:
Hey, i have a question about the adc. I'm trying to read an adc pin on a ATSAME53N20A.

# My Test code

```
#[embassy_executor::task]
pub async fn mytest() {
    let mut peripherals = Peripherals::take().unwrap();
    let pins = myboard::bsp::pin::Pins::new(peripherals.PORT);
    let _core = CorePeripherals::take().unwrap();
    let mut clocks = GenericClockController::with_external_32kosc(
        peripherals.GCLK,
        &mut peripherals.MCLK,
        &mut peripherals.OSC32KCTRL,
        &mut peripherals.OSCCTRL,
        &mut peripherals.NVMCTRL,
    );

    let mut adc1 = Adc::adc1(
        peripherals.ADC1,
        &mut peripherals.MCLK,
        &mut clocks,
        GEN_A::GCLK1,
    );

    let mut hardware_info_pin = pins.hardware_info.into_alternate::<B>();

    let data: u16 = adc1.read(&mut hardware_info_pin).unwrap();
    debug_rprintln!("Value: {}", data);
}
```

# Error
```
13:44:59.863 panicked at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/atsamd-hal-0.16.0/src/thumbv7em/adc.rs:268:1:
13:44:59.863 adc clock setup failed
```

What could be the cause for this error? Other Tasks run as expected and reading a simple GPIO also works.

_Originally posted by @pustekuchen91 in https://github.com/atsamd-rs/atsamd/issues/439#issuecomment-1833828729_

ianrrees, 2023-12-04:
Whoops, off-by-one error: it's GCLK0 that generates the the main core clock, not GCLK1...

pustekuchen91, 2023-12-04:
Thank you for splitting this up. 

Yeah GCLK1 should be fine as mentiond. I'm parallel using the microchip harmony framework and configured it like this

[image]
[image]

With the harmony code it works like intended.

pustekuchen91, 2023-12-04:
> Is GCLK1 configured in your code? Harmony will generate the setup code for you, but you need to manually setup the clock in your rust code, I believe.

@sakian No i didnt explizit configured GCLK1 in rust. I didnt for GLCK10 either.

ianrrees, 2023-12-04:
@pustekuchen91 - is it OK to close this issue, or is there more to it that is not captured by https://github.com/atsamd-rs/atsamd/issues/707 ?

pustekuchen91, 2023-12-05:
In this Issue we could take a closer look at why GLCK1 is not usable as clock.

sakian, 2023-12-05:
It is strange. Standard v1 clocking sets up GCLK1 ([code](https://github.com/atsamd-rs/atsamd/blob/3b60d03740cb679bada4fe5bbbd9e03008e9e322/hal/src/thumbv7em/clock/v1.rs#L209-L217)) then ADC tries to configure it as well, which panics since it's already configured. Maybe I'm missing something here, but I wonder how this ever worked with v1 clocking.

Solution, imo, is to move to v2 clocking, and move peripherals to support v2 properly and not configure their own clocks.

pustekuchen91, 2023-12-05:
Thank you @sakian for poiting this out. That should be the reason. 

[GLCK5 is also intialized ](https://github.com/atsamd-rs/atsamd/blob/3b60d03740cb679bada4fe5bbbd9e03008e9e322/hal/src/thumbv7em/clock/v1.rs#L224). I think this is not ideally for everyone.

closed 2024-01-10 by ianrrees