atsamd-rs/atsamd #707: Allow drivers to be used without clock configuration

pustekuchen91, 2023-12-04:
Hello,

currently i'm using a mix of harmony v3 + C++ Modules and new Modules written with rust / embassy and this crate. 
Therefore the clock configuration for many drivers like adc, spi etc. are already done by the harmony code.

It would be nice, if i can use the drivers from the atsamd crate without passing the clock configuration. because it would be done twice and could lead to problems.

Would this be possible? What needed to be changed? Maybe i can do some changes by myself and create a pull request for this.

ianrrees, 2023-12-04:
@pustekuchen91 I don't fully understand what you are proposing, however my interpretation is that you'd like to effectively mix the parts of Microchip's HAL with parts of this one, is that right?  To me, this seems opposed to the design/philosophy of this project, since we leverage features of Rust that C simply does not have; in Rust we can build a library that prevents some types of errors, like if the programmer attempts to use a peripheral that doesn't have a clock configured.

Are there specific features of the Microchip HAL's clock management that this HAL lacks?

pustekuchen91, 2023-12-05:
@ianrrees Yes, you understood correctly. I can understand your point of view. This behavior shouldnt be de default, it could maybe an feature in this crate.

Actually in the case of ADC, its very restrictive with no degree of freedom in the configuration of the clocks. What if i dont want to use `DFLL` as ClockSource?

The clock configuration in harmony has many more options and this crate lacks some features of clock configuration e. g. [clock](https://github.com/atsamd-rs/atsamd/blob/3b60d03740cb679bada4fe5bbbd9e03008e9e322/hal/src/thumbv7em/clock/v1.rs#L330) or is it implemented with v2? I didnt deep dive into it yet.

sakian, 2023-12-05:
I haven't found any limitations when configuring the clock using the v2 clocking updates. Harmony does make it easy to configure, with a nice UI, so you'll need to translate it to v2 code yourself, but I think it should be doable.

I do agree with you that there is more work to be done to properly support the v2 clocking changes in the peripheral interfaces. In fact, I already changed the way the ADC init function works in my fork of the HAL ([see commit](https://github.com/VitalBio/atsamd/commit/5bdf054cdc7c8b0325e412f8c7b4e7549c52710e)) but haven't had time to submit a PR here yet. I don't think the peripherals should be doing any clock setup, but should take a configured peripheral clock (my changes could probably do this better).

pustekuchen91, 2023-12-05:
@sakian thats looks a lot better. For the case that the clock configuration of harmony and the rust part is identical, it shouldnt matter, if its done twice (just some duplicate code).

ianrrees, 2024-08-18:
@pustekuchen91 - is the current API sufficient for your needs?  Wondering if this is OK to close out.