atsamd-rs/atsamd #722: Is USB device working with Rtic  on atsamE54 ?

yann77500, 2024-01-23:
Hello,
I tried to follow the feather_m0 usb_echo example to build a prog to atsamE54 xpro board. 
Unfortunatly cargo build failed with following issue app::usb_device::usb::UsbBus
no implement trait for atsamd_hal::usb::UsbBus

I suppose the issue is due to app made by rtic2 

Do you know a solution please? 

Thanks by advance

jbeaurivage, 2024-01-23:
Hi @yann77500, first of all, you're better off following an example from another board like `feather_m4`, because its chip (ATSAMD51) more closely resembles yours (ATSAME54), compared to a `feather_m0` which uses an ATSAMD21. They are different architectures and therefore some implementations differ. I doubt this has anything to do with the RTIC v2 framework you are using.

Furthermore, consider also joining our [matrix chat room](https://matrix.to/#/#atsamd-rs:matrix.org) where we can help you work out your issue more easily.

yann77500, 2024-02-02:
Hi @jbeaurivage , 
Thanks for your reply. 
I finally understand the origin of the issue. I think it was a conflict version between usb crates and atsamd hal.
I tried to build usbd-hid crate v 0.7.0 and atsamd 0.16.0, but only works usbd-hid v0.6.0 with atsad 0.16.0. 

Now I can build and run usb hid device with RTIC 2 environnement on ATsamE54 Xplain pro board 

Thanks for your help.

closed 2024-02-02 by yann77500