atsamd-rs/atsamd #724: edgebadge: embedded-graphics 0.8.0+ compatibility

LuckyTurtleDev, 2024-02-08:
The edgebade does still stuck on embeded-graphics 0.7.0 because of the outdated t7735-lcd-rs crate.

see also: https://github.com/LuckyTurtleDev/pybadge-high/issues/11

LuckyTurtleDev, 2024-02-08:
I currently do not have access to an edgebade for testing. Wwould it be ok to simple increase the st7735-lcd-rs  crate version and keep everything else the same?

jbeaurivage, 2024-02-08:
Hey @LuckyTurtleDev, as I understand it, you want to bump the `embedded-graphics` dependency of `edgebadge`, correct?
You can open a PR with the appropriate version bumps, and we will review it.

LuckyTurtleDev, 2024-02-15:
> Hey @LuckyTurtleDev, as I understand it, you want to bump the `embedded-graphics` dependency of `edgebadge`, correct? You can open a PR with the appropriate version bumps, and we will review it.

Yes, I planed it for today. But it looks like @orsinium was faster.

PR #726 (2024-02-15): edgebadge: update embedded-graphics to 0.8
# Summary
fix #724 

# Checklist
  - [x] `CHANGELOG.md` for the BSP or HAL updated
  - [x] All new or modified code is well documented, especially public items
  - [x] No new warnings or clippy suggestions have been introduced - CI will **deny** clippy warnings by default! You may `#[allow]` certain lints where reasonable, but ideally justify those with a short comment.
closed 2024-02-15 by jbeaurivage