atsamd-rs/atsamd #729: Neo Trinkey -- Capacitive touch pad support

lynnpepin, 2024-03-23:
Per [https://github.com/atsamd-rs/atsamd/pull/471], the Adafruit Neopixel Trinkey support was merged in without support for the capacitive touch buttons.

I'm happily using `neo_trinkey` without it :) But I am personally looking for such support.

I would actually love to take this on myself. I'm a novice in both Rust and embedded, and I've scoured the existing Rust and CircuitPython source but I'm a bit lost. If anyone has any pointers or recommendations, I'd certainly appreciate it so I can poke at it during weekends :)

sajattack, 2024-03-23:
there's more details on the peripheral touch controller in #420

lynnpepin, 2024-03-23:
Checking it out-- thank you!

ianrrees, 2025-03-05:
@lynnpepin - I'm closing this out as there hasn't been any activity in close to a year, but please feel free to open an issue or PR if you get back to it!  Several of us hang out [on Matrix](https://matrix.to/#/#atsamd-rs:matrix.org) too, if realtime chat is preferable.

closed 2025-03-05 by ianrrees